Koenkk/zigbee2mqtt #27817: Nous E5 no battery reporting

nmaster2042, 2025-06-26:
### What happened?

I added this temp / hum sensor on my zigbee2mqtt.
All is fine Except there is no battery report.

There is a ?

Tried to remove and add on zigbee network several times.

[image]

[image]

### What did you expect to happen?

Device shoud send battery % reporting.

### How to reproduce it (minimal and precise)

Just pair the Nous E5 on the zigbee network.

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

ZStack3x0 20230507

### Adapter

SONOFF dongle -P

### Setup

Jeedom jeezigbee plugin

nmaster2042, 2025-07-02:
Tried after upgrading to 2.5.0, same.

nmaster2042, 2025-09-02:
> Not fixed in version 2.6.1

I confirm after upgrading to 2.6.1

phi-lefev, 2025-09-03:
Here is the result :

[image]

[image]

Passific, 2025-09-03:
It worked on my side. I’m waiting to see if it stays.

phi-lefev, 2025-09-03:
It works for me now too.
I hadn't woken up the sensor properly.
Thank you very much for this solution.