Koenkk/zigbee2mqtt #27821: Frequent reboots (waiting for response TIMEOUT)

Ivano62, 2025-06-26:
### What happened?

After the penultimate DEV update, it often happens, many times every day, that it restarts

### What did you expect to happen?

No reboots

### How to reproduce it (minimal and precise)

Install the penultimate DEV update

### Zigbee2MQTT version

2.4.0-dev commit: 4190ba43b43e1f688e246f9236bfe7c60230a927

### Adapter firmware version

0x26720700

### Adapter

RaspBee2

### Setup

Linux raspberryha 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
Raspbian GNU/Linux 10
Docker container

### Debug log

Every restart:

/app/node_modules/.pnpm/zigbee-herdsman@4.2.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:691
            req.reject(new Error("waiting for response TIMEOUT"));
                       ^
Error: waiting for response TIMEOUT
    at DeconzAdapter.checkWaitForDataRequestTimeouts (/app/node_modules/.pnpm/zigbee-herdsman@4.2.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:691:24)
    at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@4.2.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:87:18)
    at listOnTimeout (node:internal/timers:588:17)
    at processTimers (node:internal/timers:523:7)

Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.

Koenkk, 2025-06-27:
@manup this is likely caused by https://github.com/Koenkk/zigbee-herdsman/pull/1417

@Ivano62 could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

manup, 2025-06-27:
Hi I could reproduce this error in my setup  will create a fix PR asap.

Koenkk, 2025-06-27:
@manup I'm not sure how many people are affected by this, but keep in mind a new z2m release is planned for Tuesday. If many people are going to be affected by this it might be better to temporarily revert https://github.com/Koenkk/zigbee-herdsman/commit/c719ee45629f52132f0b673cecd5a4491478ee44

manup, 2025-06-27:
@Koenkk with the fix in #1432 it should be good to go.

Tested in my setup where I could reproduce this with a device which was in the network but not known to z2m yet (the error could have occurred in other cases too).

Classic race condition.

```
[2025-06-27 21:31:05] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 45 srcAddr: 0x6552 destAddr: 0x0 profile id: 0x0104 cluster id: 0x0006 lqi: 255
[2025-06-27 21:31:05] debug: 	zh:controller: Trying to identify unknown device with address '25938'
[2025-06-27 21:31:05] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x6552 EP:0 seq: 46 requestId: 1
[2025-06-27 21:31:05] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 1 status: Success
[2025-06-27 21:31:15] debug: 	zh:deconz: Timeout for request in openRequestsQueue addr: 25938, clusterId: 8001, profileId: 0, seq: 1

/home/mpi/src/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:697
            req.reject(new Error("waiting for response TIMEOUT"));
                       ^
Error: waiting for response TIMEOUT
    at DeconzAdapter.checkWaitForDataRequestTimeouts (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:697:24)
    at Timeout._onTimeout (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:90:22)
    at listOnTimeout (node:internal/timers:608:17)
    at processTimers (node:internal/timers:543:7)
 ELIFECYCLE  Command failed with exit code 1.
```

TLTR: In the logs an IEEE address request was sent to a unknown sleeping device (Xiaomi switch), which doesn't answer. The response waiter,  timed out before the the APS confirm was received, which caused the crash.

Koenkk, 2025-06-28:
Thanks for the quick fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Ivano62, 2025-06-29:
Hi,
after more than 30 hours of installing the DEV-16faef1 version, no reboot occurred.
Now I try the latest DEV (6930fa6). 
Thanks

closed 2025-06-30 by Koenkk