Koenkk/zigbee2mqtt #27833: TS0601_water_switch

meho24, 2025-06-27:
### Link

https://www.zigbee2mqtt.io/devices/TS0601_water_switch.html#tuya-ts0601_water_switch

### Database entry

0xc02cedfffe72fa5e

### Zigbee2MQTT version

2.4.0-1

### Comments

[Valve 1 (switch, l1 endpoint)](https://www.zigbee2mqtt.io/devices/TS0601_water_switch.html#valve-1-switch-l1-endpoint)
The current state of this switch is in the published state under the state_l1 property (value is ON or OFF). To control this switch publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"state_l1": "ON"}, {"state_l1": "OFF"} or {"state_l1": "TOGGLE"}. To read the current state of this switch publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"state_l1": ""}

### External definition

```shell
Device can no be toggled on or off with the mqtt payload {"state_l1": "ON"} or {"state_l1": "OFF"}.  It will receive the valve information but you can't toggle it via zigbee2mqtt
```

### What does/doesn't work with the external definition?

Toggle of the valves doesn't work

SzymonZy, 2025-07-03:
I can confirm. Set state does not work. However feedback from device is correct.

stijnder, 2025-07-05:
I can confirm the same problem in 2.5.1, also no battery level. State is updated when pressing the buttons manually. Earlier version of z2m didn't recognize the device at all, now it does.

Device _TZE284_fhvpaltk
Coördinator type
EmberZNet
Coordinator revisie
7.4.4 [GA]
zigbee_herdsman_converters_version
23.69.0
zigbee_herdsman_version
4.3.1

Logs for turning on, get state, and turning off:

info 2025-07-05 20:27:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dual water valve', payload '{"battery":null,"countdown_l1":10,"countdown_l2":10,"linkquality":96,"state_l1":"ON","state_l2":"OFF","valve_duration_l1":10,"valve_duration_l2":11,"valve_status_l1":"idle","valve_status_l2":"idle","voltage":null}'
error 2025-07-05 20:28:01z2m: No converter available for 'get' 'state' (undefined)
info 2025-07-05 20:28:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Dual water valve', payload '{"battery":null,"countdown_l1":10,"countdown_l2":10,"linkquality":96,"state_l1":"OFF","state_l2":"OFF","valve_duration_l1":10,"valve_duration_l2":11,"valve_status_l1":"idle","valve_status_l2":"idle","voltage":null}'

stijnder, 2025-07-05:
Everything works for me with the custom convertor mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/27754#issuecomment-2988303331).
Valve on/off control and battery percentage. Also all propagated to HA.

Koenkk, 2025-07-06:
Can someone provide the debug log when toggling the switch?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

kisjeth, 2025-07-06:
> Can someone provide the debug log when toggling the switch?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

I'm trying but I'm a bit inexperienced and can't find the log for the switch. However I managed to get it working with the external.

kisjeth, 2025-07-06:
> Can someone provide the debug log when toggling the switch?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Ok, I did it. I hope this is okay.

`[2025-07-06 21:21:17] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Irrigatore/set' with data '{"valve_l1":"ON"}'
[2025-07-06 21:21:17] debug: 	z2m: Publishing 'set' 'valve_l1' to 'Irrigatore'
[2025-07-06 21:21:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Irrigatore', payload '{"battery":100,"countdown_l1":1,"countdown_l2":1,"linkquality":255,"valve_l1":"ON","valve_l2":"OFF","valve_status_l1":"idle","valve_status_l2":"idle"}'
[2025-07-06 21:21:32] debug: 	z2m: Received Zigbee message from 'Irrigatore', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":104}],"seq":17664}' from endpoint 1 with groupID 0
[2025-07-06 21:21:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Irrigatore', payload '{"battery":100,"countdown_l1":1,"countdown_l2":1,"linkquality":255,"valve_l1":"ON","valve_l2":"OFF","valve_status_l1":"manual","valve_status_l2":"idle"}'
[2025-07-06 21:21:32] debug: 	z2m: Received Zigbee message from 'Irrigatore', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":13}],"seq":17920}' from endpoint 1 with groupID 0
[2025-07-06 21:21:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Irrigatore', payload '{"battery":100,"countdown_l1":1,"countdown_l2":1,"linkquality":255,"valve_l1":"ON","valve_l2":"OFF","valve_status_l1":"manual","valve_status_l2":"idle"}'
[2025-07-06 21:21:32] debug: 	z2m: Received Zigbee message from 'Irrigatore', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":59}],"seq":18176}' from endpoint 1 with groupID 0
[2025-07-06 21:21:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Irrigatore', payload '{"battery":100,"countdown_l1":1,"countdown_l2":1,"linkquality":255,"valve_l1":"ON","valve_l2":"OFF","valve_status_l1":"manual","valve_status_l2":"idle"}'`

kisjeth, 2025-07-06:
Tomorrow. There's a storm here tonight and I don't feel like having to go out into the storm to manually block the irrigation if it stays open.

stijnder, 2025-07-06:
Here's my log with the converter: https://pastebin.com/hKrvF8g2. Now, how do I remove the converter? I renamed the file, restarted z2m, and removed and re-added the device. Still shows as "Custom working dual water valve with _l1/_l2 naming"

medovs, 2025-07-07:
> > Everything works for me with the custom convertor mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/27754#issuecomment-2988303331). Valve on/off control and battery percentage. Also all propagated to HA.
> 
> Can I please ask if you created the .js file in the root directory or how? I'm trying to figure out what I did wrong.

Hi,
To make it work, you need to create the .js file in the folder "external_converters" located in the same directory as your configuration.yaml.

In my case on an LXC container under Proxmox
/opt/zigbee2mqtt/data/external_converters

```Mon Jul 07 : 10:04 -- root@zigbee2mqtt:/opt/zigbee2mqtt/data # ls
configuration_backup_v1.yaml  configuration.example.yaml  coordinator_backup.json.old  log                   migration-3-to-4.log
configuration_backup_v2.yaml  configuration.yaml          database.db                  migration-1-to-2.log  state.json
configuration_backup_v3.yaml  coordinator_backup.json     external_converters          migration-2-to-3.log
Mon Jul 07 : 10:04 -- root@zigbee2mqtt:/opt/zigbee2mqtt/data # ls external_converters/
ts0601_valve_custom.js
Mon Jul 07 : 10:05 -- root@zigbee2mqtt:/opt/zigbee2mqtt/data #```

If you use Home Assistant with the Zigbee2MQTT add-on

/config/zigbee2mqtt/external_converters (without the data folder).

I use this in ts0601_valve_custom.js
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

module.exports = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_fhvpaltk',
        },
    ],
    model: 'TS0601_valve_custom',
    vendor: 'Tuya',
    description: 'Custom working dual water valve with _l1/_l2 naming',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.binary('valve_l1', ea.STATE_SET, 'ON', 'OFF'),
        e.binary('valve_l2', ea.STATE_SET, 'ON', 'OFF'),
        e.numeric('countdown_l1', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min'),
        e.numeric('countdown_l2', ea.STATE_SET).withValueMin(0).withValueMax(255).withUnit('min'),
        e.enum('valve_status_l1', ea.STATE, ['idle', 'manual', 'remote']),
        e.enum('valve_status_l2', ea.STATE, ['idle', 'manual', 'remote']),
        e.numeric('battery', ea.STATE).withUnit('%'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'valve_l1', tuya.valueConverter.onOff],
            [2, 'valve_l2', tuya.valueConverter.onOff],
            [13, 'countdown_l1', tuya.valueConverter.countdown],
            [14, 'countdown_l2', tuya.valueConverter.countdown],
            [104, 'valve_status_l1', tuya.valueConverterBasic.lookup({
                idle: tuya.enum(2),
                manual: tuya.enum(0),
                remote: tuya.enum(1),
            })],
            [105, 'valve_status_l2', tuya.valueConverterBasic.lookup({
                idle: tuya.enum(2),
                manual: tuya.enum(0),
                remote: tuya.enum(1),
            })],
            [59, 'battery', tuya.valueConverter.raw],
        ],
    },
};
```

medovs, 2025-07-07:
> Hi, How use 'ts0601_valve_custom.js' ; in my case, no change ;) THX

Need reboot zigbee2mqtt and the device load the new parameters.

Koenkk, 2025-07-08:
Fixed the issue!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-07-08 by Koenkk

Nepomuk3000, 2025-08-06:
Hi all,
In my case, MQTT valve commands work fine (zigbee2mqtt 2.6.0) **BUT** only a few seconds after a manual action on its buttons.
This is, of course, not the intended behavior ;-)
Do you have a trick  ?
Thanks

Nepomuk3000, 2025-08-06:
Thanks Ano10 !

Using the Zigbee2MQTT GUI, it seems to works with this procedure:

- First, set the countdown 
- Then toggle it on

I’ll let you know more after further testing.