Koenkk/zigbee2mqtt #27836: Legrand Self-e (Green Power) - No events received outside pairing mode despite successful pairing

Netatout, 2025-06-28:
Hi,

I'm experiencing an issue with Legrand Self-e Green Power switches (batteryless, no LED), integrated via Zigbee2MQTT (v2.4.0-1) running within Home Assistant using an external converter and a Sonoff Dongle-P (CC2652P) coordinator.

The pairing process completes successfully:
-	Zigbee2MQTT receives a `commissioning` message with a `srcID`.
-	The device is correctly identified as `ZLGP17/ZLGP18`.
-	Actions like `press_once` and `press_twice` are published over MQTT *but only while `permit_join` is enabled*.

However, *once `permit_join` is disabled, no events are received anymore, making the device unusable in automation.*

---

🔁 **Important context**

-	I use Zigbee2MQTT integrated in *Home Assistant (HA)*.
-	I own *multiple Legrand Self-e switches* (same model, same reference).
-	Older units (bought a few months ago) *continue to function normally outside pairing mode*.
-	A newly received unit, although identical in appearance and model ID, *stops sending events once pairing mode is off*.

I *do not have a Legrand gateway*, so I cannot check or update the firmware.
I suspect newer hardware revisions or firmware changes might be causing this behavior.

---

🧪 Observed behavior

Switch	Pairing	Events outside pairing	Works

#1 (older)	✅ OK	✅ Yes	✅
#2 (recent)	✅ OK	❌ No	❌

---

⚙️ Environment

Platform: Home Assistant OS (Zigbee2MQTT as Add-on)

Zigbee2MQTT: v2.4.0-1

Coordinator: Sonoff Dongle-P (CC2652P)

Firmware: [e.g. 20230507]

green_power: true enabled in configuration.yaml

---

❓ Questions

Could this be caused by a firmware change in newer Self-e switches, preventing srcID recognition when permit_join is off?

Is there a recommended way to manually persist or whitelist Green Power srcIDs, so they are accepted after pairing?

Thanks for your help 🙏
I can provide additional debug logs if needed.

Netatout, 2025-07-05:
Hello,

Here are the photos of the two switches. On the left is the functional one, on the right is the one that doesn't work. We can see some slight variations in some information on the front. But the reference is identical.

[image]
[image]

Nerivec, 2025-07-05:
What about the last question about firmware versions shown in frontend?

Netatout, 2025-07-07:
I don't know how to get the firmware version.
Where can I find this information?

Netatout, 2025-07-23:
Hi,

I was unable to re-pair my Legrand ZLGP17 (Green Power) switch using a Z-Stack coordinator (SLZB-06), even after full reset and unblocking the device.

Switching to a SLZB-MR3 with EZSP (Ember) firmware fixed the issue immediately.
Seems Z-Stack has trouble with Green Power pairing, while EZSP works fine.

Thanks!