Koenkk/zigbee2mqtt #27855: Ikea INSPELNING Smart plug E2206 raport Voltage, Power, Energy = 0

Jabol86, 2025-06-30:
### What happened?

I bought 3 ikea sockets. After adding to z2m on one of them all information appears, and then in two there is no data except amps.

Work:

[image]

Not work:

[image]

Soft:

[image]

### What did you expect to happen?

I wish they all worked :(

### Zigbee2MQTT version

2.4.0-1

### Adapter firmware version

20240710

### Adapter

SLZB-06 (CC2652P)

### Setup

Add-on on Home Assistant OS on gmtek g3 mini

### Debug log

[2025-06-30 22:26:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Suszarka new', payload '{"current":0.08,"energy":0,"identify":null,"linkquality":61,"power":0,"power_on_behavior":"previous","state":"ON","update":{"installed_version":33816645,"latest_version":33816645,"state":"idle"},"voltage":229.5}'

mrmaximas, 2025-06-30:
try to re-configure plugs.

Jabol86, 2025-07-01:
> try to re-configure plugs.

Try meny times ;/

mrmaximas, 2025-07-02:
Everything works for me as it has from day one.
[image]
[image]

[image]

LeoeLeoeL, 2025-07-02:
Issue appears under a very low load.
I think you cannot check that because your plug is monitoring an UPS.

LeoeLeoeL, 2025-07-30:
Today:

[image]

Conumption increases of 250 watt/h two times while instant power is 1.5 watt.

claymen, 2025-09-04:
Mine all report their instant power and voltage, just not all of them report the total consumed. Though weirdly one of the broken ones now is reporting it, and is definitely showing usage that matches the 20w or so of load that device generally uses.

Koenkk, 2025-09-24:
You can try force reading the power value via the frontend:

[image]

If it logs `zhc:tz: Read result of 'haElectricalMeasurement': {"activePower":0}` when under load it means an issue with the plug (and thus it cannot be solved from z2m)

Koenkk, 2025-09-26:
Can you try to read the currentSummDelivered? Its available in the seMetering cluster

Koenkk, 2025-10-01:
Just send it, I will filter

LeoeLeoeL, 2025-10-01:
Excuse me if i'm so straightforward but,there should be a priority to fix the problem of spikes of energy metering than this one.
Because, when you fix the issue of energy=0 then the people meet the other one.

Koenkk, 2025-10-02:
@claymen I checked the log but the device really reports `energy: 0`:

```
[2025-10-02 08:55:54] debug: 	z2m: Received Zigbee message from 'SW-Voron', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":0}' from endpoint 1 with groupID 0
```

Given the other one works, could it be that `SW-Voron` is just broken?

claymen, 2025-10-03:
> [@claymen](https://github.com/claymen) I checked the log but the device really reports `energy: 0`:
>
> ```
> [2025-10-02 08:55:54] debug: 	z2m: Received Zigbee message from 'SW-Voron', type [...]

Though to throw a spanner in the works, I just removed two of the ones having issues SW-Voron and SW-CR10sPro, reconnected my old tubeszb-cc2652-poe via ZHA and they are now reporting. Basically have both of the offending plugs plugged into each other and to a light I know draws about 45w

Interesting the wildly different voltage readings... though

[image]

[image]

But for testing, I have 4 on zigbee2mqtt (first 4) and last 2 are on ZHA

[image]

And the voltage readings across them are all different, 3 of them 250v~ the other 3 about 240v~

But all are recording total usage, and wattages are within margin of error. In any case, I'll leave them running for the day, check the totals and then re-bind the now working ones back to zigbee2mqtt and see what happens.

tompointexe, 2026-01-14:
> Just bought a fresh batch of Inspelnings. All of them seem to work OK except one, that is not exposing power nor voltage, but is exposing current correctly, even under high load. All of this is under Z2M and latest firmware for the plug itself.
> 
Hi,
That's exactly the issue I'm facing, returns Amps currently, but not voltage so no Watts consumed
Feels like they will just become simple smart plugs with no energy monitoring
Thanks

claymen, 2026-03-16:
It's such a weird issue, as I've had plugs which didn't work for some time, suddenly then start working. Trying to narrow down any changes, and the only change was introducing a few Shelly EM Mini Gen4's in Zigbee mode which I noticed seem to get a much higher preference for devices to connect to. With that said, if you are using Home Assistant you can workaround the issue with a helper, it's not perfect but at least because my wattage was always accurate, this gets close enough for general power metering.