Koenkk/zigbee2mqtt #27859: I am migrating some Smartwings WM25L-Z Rollar blinds from ZHA to Z2M, and when I add the devices they show as Mains (Single Phase) for their power, not Battery. This seems to be causeing the devices to fall offline.

moosedoggie, 2025-07-01:
### What happened?

Added devices to z2m, device shows as "Mains (Single Phase)"

### What did you expect to happen?

Device should show "Battery" 

### How to reproduce it (minimal and precise)

Add Smartwings WM25L-Z Rollar blind to z2m

### Zigbee2MQTT version

2.4.0 commit: unknown

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-MR3 (ember)

### Setup

Home assistant Add-on

moosedoggie, 2025-07-01:
```
{"id":5,"type":"EndDevice","ieeeAddr":"0x94ec32fffefdde1a","nwkAddr":62294,"manufId":4098,"manufName":"Smartwings","powerSource":"Mains (single phase)","modelId":"WM25/L-Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,258],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"WM25/L-Z","manufacturerName":"Smartwings"}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xbc8d7efffe239879","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0xbc8d7efffe239879","endpointID":1}],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":290210452},"lastSeen":1751384104137}
```

Koenkk, 2025-07-01:
Fixed! After updating make sure to re-configure the device by pressing the yellow refresh button on the frontend -> device page and restart z2m.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and in the next release which is every 1st of the month.

closed 2025-07-01 by Koenkk