Koenkk/zigbee2mqtt #27864: Not starting with a Conbee II key

jiquem, 2025-07-01:
### What happened?

I have 2 z2m in my setup. One with a SZLB and one with a Conbee 2. The one with the Conbee is no more starting following upgrade to 2.5.

### How to reproduce it (minimal and precise)

Update to 2.5. with a Conbee 2

### Zigbee2MQTT version

2.5.0

### Adapter firmware version

N/A

### Adapter

Conbee 22

### Setup

Linux a0d7b954-ssh 6.12.23-haos on Proxmox

### Debug log

[21:10:01] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-01 21:10:01] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt-conbee/log/2025-07-01.21-04-16'
[2025-07-01 21:10:01] info: 	z2m: Logging to console, file (filename: log.log)
[2025-07-01 21:10:01] debug: 	z2m: Loaded state from file /config/zigbee2mqtt-conbee/state.json
[2025-07-01 21:10:01] info: 	z2m: Starting Zigbee2MQTT version 2.5.0 (commit #unknown)
[2025-07-01 21:10:01] info: 	z2m: Starting zigbee-herdsman (4.3.0)
[2025-07-01 21:10:01] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6780,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt-conbee/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt-conbee/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt-conbee/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2679536-if00\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-07-01 21:10:01] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2679536","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2679536-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyACM1","manufacturer":"0658","pnpId":"usb-0658_0200-if00","vendorId":"0658","productId":"0200"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-07-01 21:10:01] debug: 	zh:adapter:discovery: Unable to match USB adapter: deconz | /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2679536-if00
[2025-07-01 21:10:01] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6780,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2679536-if00","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt-conbee/database.db","databaseBackupPath":"/config/zigbee2mqtt-conbee/database.db.backup","backupPath":"/config/zigbee2mqtt-conbee/coordinator_backup.json"}'
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Opening serial port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2679536-if00
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Serialport opened
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Query firmware parameters
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Reset firmware watchdog
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Send write parameter request from queue. seq: 1 parameter: DEV_WATCHDOG_TTL
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Write parameter response parameter: DEV_WATCHDOG_TTL, status: Success
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Reset firmware watchdog success
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: Send read firmware version request from queue. seq: 2
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: read firmware version response - version: 0x26580700
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: MAC_ADDRESS seq: 4
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for MAC_ADDRESS, seq: 4, status: 0, parameter: 0x00212effff09a564
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: APS_TRUST_CENTER_ADDRESS seq: 5
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for APS_TRUST_CENTER_ADDRESS, seq: 5, status: 0, parameter: 0x00212effff09a564
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: NWK_PANID seq: 6
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for NWK_PANID, seq: 6, status: 0, parameter: 6780
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: APS_USE_EXTENDED_PANID seq: 7
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for APS_USE_EXTENDED_PANID, seq: 7, status: 0, parameter: 0xdddddddddddddddd
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_CURRENT_CHANNEL seq: 8
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for STK_CURRENT_CHANNEL, seq: 8, status: 0, parameter: 11
[2025-07-01 21:10:02] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_NETWORK_KEY seq: 9
[2025-07-01 21:10:02] debug: 	zh:deconz:frameparser: Received read parameter response for STK_NETWORK_KEY, seq: 9, status: 0, parameter: <hidden>
[2025-07-01 21:10:03] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_NWK_UPDATE_ID seq: 10
[2025-07-01 21:10:03] debug: 	zh:deconz:frameparser: Received read parameter response for STK_NWK_UPDATE_ID, seq: 10, status: 0, parameter: 0
[2025-07-01 21:10:03] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: APS_CHANNEL_MASK seq: 11
[2025-07-01 21:10:03] debug: 	zh:deconz:frameparser: Received read parameter response for APS_CHANNEL_MASK, seq: 11, status: 0, parameter: 2048
[2025-07-01 21:10:03] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_PROTOCOL_VERSION seq: 12
[2025-07-01 21:10:03] debug: 	zh:deconz:frameparser: Received read parameter response for STK_PROTOCOL_VERSION, seq: 12, status: 0, parameter: 267
[2025-07-01 21:10:03] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_FRAME_COUNTER seq: 13
/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82
            result = view.getUint32(pos, littleEndian);
                          ^
RangeError: Offset is outside the bounds of the DataView
    at DataView.prototype.getUint32 (<anonymous>)
    at parseReadParameterResponse (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82:27)
    at parseFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:633:32)
    at processFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:582:53)
    at Driver.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58)
    at Driver.emit (node:events:518:28)
    at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22)
    at Parser.emit (node:events:518:28)
    at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)

steffanlm, 2025-07-01:
Got same here, slightly different errorlog - but same fault:
`21:31:08] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82
            result = view.getUint32(pos, littleEndian);
                          ^
RangeError: Offset is outside the bounds of the DataView
    at DataView.prototype.getUint32 (<anonymous>)
    at parseReadParameterResponse (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82:27)
    at parseFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:633:32)
    at processFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:582:53)
    at Driver.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58)
    at Driver.emit (node:events:518:28)
    at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22)
    at Parser.emit (node:events:518:28)
    at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)`

marcinwadon, 2025-07-01:
Same here:
```
[2025-07-01 22:17:39] debug: 	zh:deconz:driver: send read parameter request from queue. parameter: STK_FRAME_COUNTER seq: 13
/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82
            result = view.getUint32(pos, littleEndian);
                          ^
RangeError: Offset is outside the bounds of the DataView
    at DataView.prototype.getUint32 (<anonymous>)
    at parseReadParameterResponse (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82:27)
    at parseFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:633:32)
    at processFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:582:53)
    at Driver.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58)
    at Driver.emit (node:events:518:28)
    at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22)
    at Parser.emit (node:events:518:28)
    at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)
```

z-master42, 2025-07-01:
My ConBee 2 has the firmware version 26720700 and Z2M 2.5.0 runs with it without any problems.

phil-schneider, 2025-07-01:
Hello,
I am Also on Version 0x26720700
How can I Upgrade the conbee to the latest Firmware Version?

manup, 2025-07-01:
> Hello, I am Also on Version 0x26720700 How can I Upgrade the conbee to the latest Firmware Version?

Instructions can be found by following link "Flashing" at: https://www.zigbee2mqtt.io/guide/adapters/deconz.html

marcinwadon, 2025-07-01:
> > Hello, I am Also on Version 0x26720700 How can I Upgrade the conbee to the latest Firmware Version?
> 
> Instructions can be found by following link "Flashing" at: https://www.zigbee2mqtt.io/guide/adapters/deconz.html
> 

Thanks! After upgrading my ConBeeII to the newest firmware available, zigbee2mqtt started working again :)

manup, 2025-07-01:
> Hello,
> 
> If it can be helpzl, I have an issue with the RaspBee 2 to :
> 
> [23:54:35] INFO: Preparing to start... [23:54:35] INFO: Socat not enabled [23:54:36] INFO: Starting Zigbee2MQTT... Starting Zigbee2MQTT without watchdog. /app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82 result = view.getUint32(pos, littleEndian); ^ RangeError: Offset is outside the bounds of the DataView at DataView.prototype.getUint32 () at parseReadParameterResponse (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82:27) at parseFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:633:32) at processFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:582:53) at Driver. (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58) at Driver.emit (node:events:518:28) at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22) at Parser.emit (node:events:518:28) at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14) at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)

It's the same error as above comments, the PR https://github.com/Koenkk/zigbee-herdsman/pull/1438 provides a fix for older firmware versions.

SCUZNUTS, 2025-07-02:
Same issue on Conbee II, updated to 0x26780700.
'Worked' on 2.5.0, but was unstable. Ill refrain from posting error logs for now.
Rolling back to 2.4.0 and is stable.
If the next version is similar unstable on 0x26780700 ill open an issue too.

SmTo81, 2025-07-02:
Hi I also have another problem: 
[08:11:36] INFO: Starting Zigbee2MQTT...
node:internal/modules/run_main:91
  const type = getNearestParentPackageJSONType(mainPath);
               ^
Error: Invalid package config /app/package.json.
    at shouldUseESMLoader (node:internal/modules/run_main:91:16)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:163:20)
    at node:internal/main/run_main_module:36:49 {
  code: 'ERR_INVALID_PACKAGE_CONFIG'
}
Node.js v22.15.1

gnilebein, 2025-07-02:
> Hi I also have another problem: [08:11:36] INFO: Starting Zigbee2MQTT... node:internal/modules/run_main:91 const type = getNearestParentPackageJSONType(mainPath); ^ Error: Invalid package config [...]

If you have a different problem, you might want to open a different issue ;-)

gromitn, 2025-07-02:
Same issue - Z2M won't start on latest version.

`RangeError: Offset is outside the bounds of the DataView
    at DataView.prototype.getUint32 (<anonymous>)
    at parseReadParameterResponse (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:82:27)
    at parseFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:633:32)
    at processFrame (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:582:53)
    at Driver.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58)
    at Driver.emit (node:events:518:28)
    at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22)
    at Parser.emit (node:events:518:28)
    at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)
    at Driver.onParsed (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1475:22)
    at Parser.emit (node:events:518:28)
    at Parser.onMessage (/app/node_modules/.pnpm/zigbee-herdsman@4.3.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)`

Cissoubaka, 2025-07-02:
I updated my fw to the last one ( 0x26780700)
but having this error :
`[2025-07-02 10:07:36] debug:     zh:adapter:discovery: Unable to match USB adapter: deconz | /dev/ttyUSB0`

manup, 2025-07-02:
Note when updating the firmware please keep the ConBee connected for more than 10 seconds **after** the firmware was flashed so it can boot up once. If it is disconnected from USB port to quickly the update may not be applied.

Cissoubaka, 2025-07-02:
that was it, thanks

roofburner, 2025-07-02:
ConBee2/RaspBee2 USB stick

Firmware upgrade from 0x26580700 to 0x26780700

zigbee2mqtt 2.5.0 now starts. Hope it runs stable.

Thanks for all comments!

Koenkk, 2025-07-02:
I will make a hot fix release (2.5.1) with a fix for it included (thanks for the fast PR @manup!)

fredhohenstein, 2025-07-02:
Hi Erwin. Every day I make a full backup of my two ha boxes. Backup created IN ha. Beforer evey addon update I also create a backup
 This addon backup, created before upgrade to 2
5.0 was restored

Googeling I found this info to restore the backup

To restore a Home Assistant add-on backup, you'll first need to create a snapshot (backup) of your Home Assistant instance, including the add-ons. Then, when restoring, you can choose to restore specific add-ons or all of them. If you're restoring to a new instance, you may need to manually install some add-ons before restoring the backup. 
Detailed Steps:
1. Create a Backup:
Go to Settings > System > Backups in your Home Assistant interface.
Click Create Backup and choose the type of backup (full or partial). A full backup includes everything, while a partial backup allows you to select specific components.
For add-on restoration, ensure the add-ons you want to restore are included in the backup.
Wait for the backup to complete.
2. Restore the Backup:
Navigate back to Settings > System > Backups.
Select the backup you want to restore.
If it's a partial backup, select the specific add-ons you want to restore.
Enter your restore password (if required).
Initiate the restore process.
3. Considerations for Add-on Restoration:
Fresh Installation: If restoring to a new Home Assistant instance, you might need to manually install some add-ons (like those not included in the official store) before restoring the backup.
Sequential Restoration: If restoring add-ons individually, install them in sequence, then restore from your backup. 
Full Reboot: After restoring a full backup, it's recommended to do a full reboot of your Home Assistant OS (not just a restart) to ensure all services are properly initialized. 
4. Restoring Data Only:
If you need to restore only the data of a specific add-on, you might need to extract it from the backup file manually. 
Important Notes:
Backups created through the Home Assistant interface include add-on data. 
The restore process can take some time, especially for larger installations. 
If you encounter issues with add-ons not starting after a restore, try a full reboot or check if they need to be reinstalled. 
For add-ons not in the official store, manual installation might be necessary before restoring the backup. 
If you are restoring to a new instance, ensure the necessary add-ons are installed first before restoring from a snapshot. 

After the restore AND a full reboot zigbee2mqtt worked normal no data loss

Succes

earvdl, 2025-07-02:
Thank you @Koenkk for the quick fix. My home is life again.

Best regards, fijne avond,
Erwin

terierul, 2025-07-03:
Hi for me is not working

The error page

[image]

Configuration of add on

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: password
serial:
  port: /dev/ttyACM0
  adapter: deconz

Conbee 2 in hardware

Subsystem:
tty
Device path:
/dev/ttyACM0
ID:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2286924-if00
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2286924-if00
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usbv2-0:1.1:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: ConBee_II
ID_MODEL_ENC: ConBee\x20II
ID_MODEL_ID: '0030'
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_1_1_0
ID_PATH_WITH_USB_REVISION: platform-fd500000.pcie-pci-0000:01:00.0-usbv2-0:1.1:1.0
ID_REVISION: '0100'
ID_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2286924
ID_SERIAL_SHORT: DE2286924
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: ConBee_II
ID_USB_MODEL_ENC: ConBee\x20II
ID_USB_MODEL_ID: '0030'
ID_USB_REVISION: '0100'
ID_USB_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2286924
ID_USB_SERIAL_SHORT: DE2286924
ID_USB_TYPE: generic
ID_USB_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_USB_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_USB_VENDOR_ID: 1cf1
ID_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_VENDOR_ID: 1cf1
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '3166984'

terierul, 2025-07-03:
added to configuration:

```
availability:
    enabled: true
    active:
        # Time after which an active device will be marked as offline in minutes (default: 10 minutes)
        timeout: 10
        # Maximum jitter (in msec) allowed on timeout to avoid availability pings trying to trigger around the same time (default: 30000, min: 1000)
        max_jitter: 30000
        # Enable timeout backoff on failed availability pings (default: true)
        # Pattern used: x1.5, x3, x6, x12... (with default timeout of 10min: 10, 15, 30, 60, 120...)
        backoff: true
        # Pause availability pings when backoff reaches over this limit until a new Zigbee message is received from the device. (default: 0, min: 0)
        # A value of zero disables pausing, else see `backoff` pattern above.
        pause_on_backoff_gt: 0
    passive:
        # Time after which a passive device will be marked as offline in minutes (default: 1500 minutes aka 25 hours)
        timeout: 1500

```

also added 
/dev/serial-by-id which I understood is preffered .... same result

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: qazwsx123
serial:
  **port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2286924-if00**
  adapter: deconz
availability:
  enabled: true
  active:
    timeout: 10
    max_jitter: 30000
    backoff: true
    pause_on_backoff_gt: 0
  passive:
    timeout: 1500
```

terierul, 2025-07-03:
moving the topic to https://github.com/Koenkk/zigbee2mqtt/issues/27889 as this one is apparently closed

derkorte, 2025-07-03:
I updated the firmware of the ConBee II to 0x26780700, but there are numerous errors in the log. Some devices are unresponsive or respond with a delay.
Here is my log:

`[2025-07-03 21:49:44] error: 	z2m: Publish 'set' 'state' to 'Licht Kellerflur ' failed: 'Error: ZCL command 0xa4c138c6c14d808b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (22) Error: Command ApsDataRequest failed with status: Busy)'
[2025-07-03 21:49:44] error: 	z2m: Publish 'set' 'state' to 'Doppelschalter Keller ' failed: 'Error: ZCL command 0xa4c1388ba05e6934/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (23) Error: Command ApsDataRequest failed with status: Busy)'
[2025-07-03 21:49:44] error: 	z2m: Publish 'set' 'state' to 'Licht Gartenzimmer ' failed: 'Error: ZCL command 0xa4c1383d93f41915/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (24) Error: Command ApsDataRequest failed with status: Busy)'
[2025-07-03 21:50:41] error: 	z2m: Failed to configure 'Steckdose 4', attempt 1 (Error: Bind 0xa4c1380f62782395/1 haElectricalMeasurement from '0x02124b001cdcaab7/1' failed (Status 'INVALID_EP')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:583:23)
    at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:40:9)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:6681:13)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[21:51:20] INFO: Preparing to start...
[21:51:20] INFO: Socat not enabled
[21:51:20] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-03 21:51:38] warning: 	z2m: Failed to ping '0xa4c138ede556ae4f' (attempt 1/1, ZCL command 0xa4c138ede556ae4f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (7) Error: Failed APS-DATA.request with confirm status: NwkRouteDiscoveryFailed (0xd0)))
[2025-07-03 21:52:39] error: 	z2m: Publish 'set' 'state' to 'Doppelschalter Keller ' failed: 'Error: ZCL command 0xa4c1388ba05e6934/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (9) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))'
[2025-07-03 21:52:44] error: 	zh:controller:device: Read response to 0xa4c138fe33767b98 failed (ZCL command 0xa4c138fe33767b98/1 genTime.readRsp({"localTime":804894764}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":197,"writeUndiv":false}) failed (failed to send ZCL request (197) Error: Command ApsDataRequest failed with status: Busy))
[2025-07-03 21:52:44] error: 	zh:controller:device: Read response to 0xa4c138fe33767b98 failed (ZCL command 0xa4c138fe33767b98/1 genTime.readRsp({"localTime":804894764}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":197,"writeUndiv":false}) failed (failed to send ZCL request (197) Error: Command ApsDataRequest failed with status: Busy))
[2025-07-03 21:52:44] error: 	zh:controller:device: Read response to 0xa4c138fe33767b98 failed (ZCL command 0xa4c138fe33767b98/1 genTime.readRsp({"localTime":804894764}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":197,"writeUndiv":false}) failed (failed to send ZCL request (197) Error: Command ApsDataRequest failed with status: Busy))
[2025-07-03 21:52:44] error: 	zh:controller:device: Read response to 0xa4c138fe33767b98 failed (ZCL command 0xa4c138fe33767b98/1 genTime.readRsp({"localTime":804894765}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":197,"writeUndiv":false}) failed (failed to send ZCL request (197) Error: Command ApsDataRequest failed with status: Busy))
[2025-07-03 21:52:44] error: 	zh:controller:device: Read response to 0xa4c138fe33767b98 failed (ZCL command 0xa4c138fe33767b98/1 genTime.readRsp({"localTime":804894765}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":197,"writeUndiv":false}) failed (failed to send ZCL request (197) Error [... cut ...]
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-07-03 22:07:43] error: 	zh:controller:device: Handling of poll check-in from 0xa4c1383f8ab6b492 failed (ZCL command 0xa4c1383f8ab6b492/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (failed to send ZCL request (26) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7)))`

dajomas, 2025-07-21:
In addition, I frequently get this message:
```
zh:controller:device: Device '0x................' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23)
```

After I interview a device that has gone offline. After that, the device usually comes back anyway