Koenkk/zigbee2mqtt #27883: Lot of errors "Error: Command ApsDataRequest failed with status: Busy"

nbranquinho, 2025-07-02:
### What happened?

After update to version 2.5.0, the system start giving a lot of errors and I don't let me control the devices in Zigbee2MQTT.
I updated to v2.5.1 but still happen

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

0x26720700

### Adapter

ConBee_II

### Setup

I'm running as ADDon in HA

### Debug log

[45df7312_zigbee2mqtt_2025-07-02T21-46-44.714Z.log](https://github.com/user-attachments/files/21026076/45df7312_zigbee2mqtt_2025-07-02T21-46-44.714Z.log)

manup, 2025-07-03:
Busy status means the APS queue is full. This happens when there are too many requests to non responsive device which due the newly always enabled ACKs can take time to process until they timeout (e.g. offline devices or devices for which there is no route yet).

Can you try to unplug the ConBee and re-attach after 10 seconds to see if that helps?

nbranquinho, 2025-07-03:
> Busy status means the APS queue is full. This happens when there are too many requests to non responsive device which due the newly always enabled ACKs can take time to process until they timeout [...]

ok, thx, I'll do your suggestion and later I leave feedback with results.

Koenkk, 2025-07-04:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.