Koenkk/zigbee2mqtt #27885: Add-One wont start / crash from time to time

phil-schneider, 2025-07-03:
### What happened?

Hello,
everything was working fine with 2.4.x and my Conbee2 stick
I also waited with the update to 2.5.0 and checked the firmware before updating.

After the update everything seems to be fine, also after a restart.
But then I noticed that the addon was suddenly not running anymore.
The logs showed what I added to debug log.

After inserting the USB stick again, it worked again.

This morning my zigbee curtain was not working.. so I restarted the addon and it worked.
10 Minutes later the add one was stopped again and I was not able to restart it with the error (again) above.

What am I doing wrong?

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

latest 

### Adapter

Conbee2

### Setup

Home Assistant

### Debug log

[07:21:09] INFO: Preparing to start...
[07:21:09] INFO: Socat not enabled
[07:21:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-03 07:21:10] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-03 07:21:10] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-03 07:21:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-03 07:21:10] error: 	z2m: Exiting...
[2025-07-03 07:21:11] error: 	z2m: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:211:28)
    at SerialPort._error (/app/node_modules/.pnpm/@serialport+stream@13.0.0/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/.pnpm/@serialport+stream@13.0.0/node_modules/@serialport/stream/dist/index.js:118:18
[07:22:14] INFO: Preparing to start...
[07:22:14] INFO: Socat not enabled
[07:22:15] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-03 07:22:36] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-03 07:22:36] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-03 07:22:36] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-03 07:22:36] error: 	z2m: Exiting...
[2025-07-03 07:22:36] error: 	z2m: undefined
TypeError: Cannot read properties of undefined (reading 'includes')
    at Controller.start (/app/lib/controller.ts:113:42)
    at start (/app/index.js:149:5)

phil-schneider, 2025-07-03:
After I unplugged the USB device and restarted everything it was working for almost an hour.
Here are the logs

[21:22:02] INFO: Preparing to start...
[21:22:02] INFO: Socat not enabled
[21:22:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-03 21:22:59] error: 	z2m: Failed to configure '0xa4c1388ee8c146fd', attempt 1 (Error: Bind 0xa4c1388ee8c146fd/1 msTemperatureMeasurement from '0xbc026efffe36542f/1' failed (failed to send ZDO request seq: (11) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))
    at DeconzAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:318:19)
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:580:30)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:221:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-07-03 21:23:11] error: 	zh:controller:device: Read response to 0x54ef441000783411 failed (ZCL command 0x54ef441000783411/1 genTime.readRsp({"timeZone":7200,"time":804885776}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":77,"writeUndiv":false}) failed (failed to send ZCL request (77) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7)))
[2025-07-03 21:24:11] error: 	zh:controller:device: Read response to 0x54ef441000783411 failed (ZCL command 0x54ef441000783411/1 genTime.readRsp({"timeZone":7200,"time":804885836}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":78,"writeUndiv":false}) failed (failed to send ZCL request (78) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7)))
[2025-07-03 21:25:23] error: 	z2m: Failed to configure '0xa4c1388ee8c146fd', attempt 2 (Error: Bind 0xa4c1388ee8c146fd/1 msTemperatureMeasurement from '0xbc026efffe36542f/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:583:23)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:221:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-07-03 21:26:03] error: 	z2m: Failed to configure '0xa4c1388ee8c146fd', attempt 3 (Error: Bind 0xa4c1388ee8c146fd/1 msTemperatureMeasurement from '0xbc026efffe36542f/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:583:23)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:221:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-07-03 21:27:50] error: 	z2m: EventBus error 'OTAUpdate/deviceMessage': CommandResponse 0xa4c138bd64890f1b/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (20) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))
[2025-07-03 21:27:50] error: 	zh:controller:device: Read response to 0xa4c138bd64890f1b failed (ZCL command 0xa4c138bd64890f1b/1 genBasic.readRsp({"appVersion":211}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":19,"writeUndiv":false}) failed (failed to send ZCL request (19) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7)))
[2025-07-03 21:35:21] error: 	zh:controller:device: Read response to 0x54ef4410007f543a failed (ZCL command 0x54ef4410007f543a/1 genTime.readRsp({"timeZone":7200,"time":804886506}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":3,"writeUn [... cut ...]
[2025-07-03 21:59:03] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-03 21:59:03] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-03 21:59:03] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-03 21:59:03] error: 	z2m: Exiting...
[2025-07-03 21:59:03] error: 	z2m: undefined
TypeError: Cannot read properties of undefined (reading 'includes')
    at Controller.start (/app/lib/controller.ts:113:42)
    at start (/app/index.js:149:5)`

`

drhirn, 2025-07-04:
Same "Cannot read properties of undefined (reading 'includes')" error here.
After the update to 2.5.1 my PTM125 switches stopped working. So i used the occasion to update the firmware of my ConBee II to 0x26780700. After that, I get the same error when starting the docker container. But, I do not know, if this error would have been there to with restarting the container without the firmware update.

Zigbee2Mqtt 2.5.1 Docker
ConBee II (Firmware 0x26780700)

Log:
```
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     z2m: Removing old log directory '/app/data/log/2025-07-04.10-29-38'
zigbee2mqtt  | [2025-07-04 10:43:29] info:      z2m: Logging to console, file (filename: log.log)
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     z2m: Loaded state from file /app/data/state.json
zigbee2mqtt  | [2025-07-04 10:43:29] info:      z2m: Starting Zigbee2MQTT version 2.5.1 (commit #5063c7d3ad88f9ed651dc36fd9e583ff452e18da
zigbee2mqtt  | )
zigbee2mqtt  | [2025-07-04 10:43:29] info:      z2m: Starting zigbee-herdsman (4.3.1)
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2465757","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2465757-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"}]
zigbee2mqtt  | [2025-07-04 10:43:29] info:      zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2465757","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2465757-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:deconz:driver: Opening serial port: /dev/ttyACM0
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:deconz:driver: Serialport opened
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:deconz:driver: Query firmware parameters
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:deconz:driver: Reset firmware watchdog
zigbee2mqtt  | [2025-07-04 10:43:29] debug:     zh:deconz:driver: Send write parameter request from queue. seq: 1 parameter: DEV_WATCHDOG_TTL
zigbee2mqtt  | [2025-07-04 10:43:31] debug:     zh:deconz:driver: tx timeout for cmd: 0b, seq: 1
zigbee2mqtt  | [2025-07-04 10:43:39] debug:     zh:deconz:driver: Send read firmware version request from queue. seq: 2
zigbee2mqtt  | [2025-07-04 10:43:39] debug:     zh:deconz:driver: Reset firmware watchdog failed
zigbee2mqtt  | [2025-07-04 10:43:41] debug:     zh:deconz:driver: tx timeout for cmd: 0d, seq: 2
zigbee2mqtt  | [2025-07-04 10:43:49] error:     z2m: Error while starting zigbee-herdsman
zigbee2mqtt  | [2025-07-04 10:43:49] error:     z2m: Failed to start zigbee-herdsman
zigbee2mqtt  | [2025-07-04 10:43:49] error:     z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
zigbee2mqtt  | [2025-07-04 10:43:49] error:     z2m: Exiting...
zigbee2mqtt  | [2025-07-04 10:43:49] error:     z2m: undefined
zigbee2mqtt  | TypeError: Cannot read properties of undefined (reading 'includes')
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:113:42)
zigbee2mqtt  |     at start (/app/index.js:149:5)
zigbee2mqtt exited with code 0
```