Koenkk/zigbee2mqtt #27886: Nous Modell B3Z - reporting of power delayed after upgrade to 2.5.x

phil-schneider, 2025-07-03:
### What happened?

Hello,
I noticed that after the upgrade to version 2.5.1 the reporting of the power sensor of my Nous B3Z is really delayed [B3Z](https://www.zigbee2mqtt.io/devices/B3Z.html#nous-b3z).
I use this to measure power of my solar panel. Before the upgrade the value of the power changed like every second.
Now it is changing like every 10 minutes, if even so

### What did you expect to happen?

I expected the power value to be updated when the values changes, it should be like every second 

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Home Assistant with addon

### Debug log

error 2025-07-03 07:51:53z2m: Publish 'get' 'power_on_behavior' to '0xa4c13831fb7d65d3' failed: 'Error: ZCL command 0xa4c13831fb7d65d3/1 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (15) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))'
error 2025-07-03 07:51:56z2m: Publish 'get' 'state' to '0xa4c13831fb7d65d3' failed: 'Error: ZCL command 0xa4c13831fb7d65d3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (17) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))'
error 2025-07-03 07:51:58z2m: Publish 'get' 'switch_type' to '0xa4c13831fb7d65d3' failed: 'Error: ZCL command 0xa4c13831fb7d65d3/1 manuSpecificTuya_3.read(["switchType"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (16) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))'
error 2025-07-03 07:51:58z2m: Publish 'get' 'state' to '0xa4c13831fb7d65d3' failed: 'Error: ZCL command 0xa4c13831fb7d65d3/2 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (failed to send ZCL request (18) Error: Failed APS-DATA.request with confirm status: ApsNoAck (0xa7))'

manup, 2025-07-03:
The ApsNoAck status (0xA7) can indeed be seen more with the new adapter since APS ACKs are now enabled as default, which the old adapter didn't. Personally I'm not in favor of this since this decision shouldn't be done by the driver but the controller on a per device basis. See my comments in https://github.com/Koenkk/zigbee-herdsman/pull/1417#issue-3121175672

A big problem with APS ACKs is that they can fill up the queues easily for non reachable devices. This is a problem regardless of the coordinator hardware, any queue will eventually fill up and block if number of non reachable device + requests is high enough.

Not sure how to tackle this: Either 1) disable (old adapter) / enable (new version) APS ACks globally which is both sub optimal. Or 2) the adapter/driver starts tracking problematic devices and enables them for hard to reach devices, but ideally this shouldn't be done by the driver but the controller.

manup, 2025-07-03:
Another perhaps less invasive option is to emulate the APS ACKs in the driver so the firmware never blocks the queues needlessly 🤔

Nerivec, 2025-07-03:
deconz got a major rework in 2.5.0, so undoubtedly some kinks to iron out from wider adoption.

See https://www.zigbee2mqtt.io/guide/adapters/#recommended if you can't wait.

phil-schneider, 2025-07-22:
This issue can be closed.
I dont have a conbee Stick anymore.
With a New SLZB-06 coordinator the reporting is much more accurate

closed 2025-07-22 by phil-schneider