Koenkk/zigbee2mqtt #27888: EnOcean PTM 215Z are not working under 2.5.1.1

GunterAv, 2025-07-03:
### What happened?

I have 11 EnOcean PTM 215Z in my home.
With Version 2.4-0-1 all works fine.

under 2.51 AND 2.5.1.1 no EnOcean PTM 215Z is working

### What did you expect to happen?

that they work switches

### How to reproduce it (minimal and precise)

install Version 2.5.1.1

### Zigbee2MQTT version

2.5.1.1

### Adapter firmware version

26780700

### Adapter

Conbee II

### Setup

...

### Debug log

...

tim-dalkiewicz, 2025-07-03:
I have the same problem.

After updating to 2.5.1 only these [PTM_215Z](https://www.zigbee2mqtt.io/devices/PTM_215Z.html) aren't working anymore. Zigbee2Mqtt said its connected, but I don't see any loggings or (error) messages.
I have 6 Switches... none of them is working.

Other Switches like [TH-S04D](https://www.zigbee2mqtt.io/devices/TH-S04D.html) or [Hue dimmer switch](https://www.zigbee2mqtt.io/devices/324131092621.html) are working

My setup:
Raspi 4: HAS OS, Home Assistant
Core: 2025.7.0
Supervisor: 2025.06.2
Operating System: 15.2

Conbee Stick II

manup, 2025-07-03:
I'm currently looking into it, my Friends of Hue switch which seems to work also says it is a "PTM 215Z" but it might be different from other EnOcen modules and use different commands.

[image]

In my logs I see that there are a few commands skipped while parsing with an error.
I'll make a PR to fix these to parse any ZGP command.

Still wonder what is the difference between ZGP handling in old vs new adapter since ZGP code wasn't changed. Might be a side effect.

Nerivec, 2025-07-03:
@tim-dalkiewicz can you provide a `debug` log of trying to press buttons (even if you see nothing, so we have some context)? And also of trying to re-pair one of the switches (make sure to confirm the channel at the end BEFORE pressing any other button - see [docs](https://www.zigbee2mqtt.io/devices/PTM_215Z.html#pairing))?

phil-schneider, 2025-07-03:
> I had the same issue. Rolled back to 2.4.0-1 in HA with Conbee II and it is working again.
> 
> Nevertheless I had to restore also the configuration.yaml and the database.db files from the Z2Ms homeassistant configuration folder. After the Upgrade to 2.5.0-1 or 2.5.1-1 the device is displayed as online, but it does not receive anymore updates. Unfortunately I do not have a debug log of the device after the upgrade and because of the Womans Acceptance Factor, I currently don't want to upgrade again.
> 

@KapUttyy How did you do that?
I tried to rollback to 2.4.0, but it didn't work.  I think its because of the missions configuration and DB.

tim-dalkiewicz, 2025-07-03:
> [@tim-dalkiewicz](https://github.com/tim-dalkiewicz) can you provide a `debug` log of trying to press buttons (even if you see nothing, so we have some context)? And also of trying to re-pair one of [...]

Logfile from today and my configuration.
All PTM 215Z Topics starting with `button.buschjaeger.4er.`
I will do the re-pairing, later.

[log.txt](https://github.com/user-attachments/files/21046086/log.txt)

[configuration.txt](https://github.com/user-attachments/files/21046116/configuration.txt)

manup, 2025-07-03:
Yes it is caused by an out of range access into an array.

I put up a PR to fix it soon.

Oddly while the ZGP parsing code didn't change, the lower level received buffer was refactored in the new adapter to not yield over sized buffers to the parser — e.g. if the firmware sends a 64 byte buffer don't push a 128 byte buffer to the parser.

The ZGP parser in old and new adapter is a bit flaky and calculates some offsets under wrong assumptions with fixed magic numbers. This is what the PR will fix.

Nerivec, 2025-07-03:
That error is specific to trying to `writeXYZ` with a value out of the range of _XYZ_ (here `-2` to `UInt8`).
Based on the logs, it's the `emit` that throws, so, looks like it's the GPD header in `checkReceivedGreenPowerIndication` that can't be computed properly (one of the `writeUInt8` is being given `-2` as value, so, either `seqNr` or `commandFrameSize`, probably the latter - if `view.byteLength === 22` that would put `commandFrameSize` at `-2`).
https://github.com/Koenkk/zigbee-herdsman/blob/a27fe889c79bd50e6eb84f4961320aa3e25c02be/src/adapter/deconz/adapter/deconzAdapter.ts#L677-L686

KapUttyy, 2025-07-04:
> [@KapUttyy](https://github.com/KapUttyy) How did you do that? I tried to rollback to 2.4.0, but it didn't work. I think its because of the missions configuration and DB.

I downloaded my Full Daily Backup from before updating to 2.5.0-1.
Then I stopped Z2M and rolled back to the Z2M AddOn from the Full Daily Backup mentioned earlier. From the downloaded package I took the /homeassistant/zigbee2mqtt folder and replaced the configuration.yaml and the database.db file.
After that I restarted Z2M AddOn which then came up with a working FoH Switch.

KapUttyy, 2025-07-08:
So as I can see the issue should be fixed with the already released zigbee-herdsman version, right?
Using Z2M as a homeassistant AddOn, I still have to wait for a new Z2M version as herdsman is part of the AddOn, correct?

manup, 2025-07-10:
> I've updated to 2.5.1-1 The switches still do not work - do I have to repair?

No, the fix is in a version after that. Currently in dev channel. See https://github.com/Koenkk/zigbee2mqtt/issues/27888#issuecomment-3047878732

RayLio42, 2025-07-21:
Hi, thanks for the fix, but when will we see it in prod release?
The wife acceptance factor is under a stress test right now as no wall switch works anymore…

tzitzaki, 2025-07-22:
Guess based on prior releases: starting of each month.

There is a section in the readme how to switch to the edge version of the plugin relatively hassle-free.
Linked above.

And as always, stick to the etiquette; no asking for ETA 😘