Koenkk/zigbee2mqtt #27890: Cannot join any new devices to Z2M

Andrewkha, 2025-07-03:
### What happened?

Starting certain point of time (not sure when it happened) I'm no longer able to join new devices to my z2m instance. I use SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P coordinator with latest FW, also use the latest version of Z2M docker container. Exsisting devices that I joined previously work with no issues. More of that, I have another instance of z2m with the same setup, and it also doesn't join new devices. I also tried to install another instance from scratch, also no luck. When I hold a button on the device to join the Zigbee network, nothing happens... I set log level of this new instance to 'debug' but didn't find anything meaningful, maybe I just can't read it properly... 

### What did you expect to happen?

I would expect that I can join new Zigbee devices to my Zigbee network

### How to reproduce it (minimal and precise)

Very simple... Just try to join a new device

### Zigbee2MQTT version

 2.5.1 commit: 5063c7d3ad88f9ed651dc36fd9e583ff452e18da

### Adapter firmware version

 20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Z2M is running in a docker container on QNAP NAS based on x86 architectrue

[andrewkha@DachaNAS ~]$ uname -a && cat /etc/issue.net
Linux DachaNAS 5.10.60-qnap #1 SMP Mon May 26 01:07:55 CST 2025 x86_64 GNU/Linux
cat: /etc/issue.net: No such file or directory

### Debug log

For simplicity, I'm attaching the log and config files from the new Installation I tried to create from scratch. If required I can do the same for my production one. But the behaviour is completely identical

[log.log](https://github.com/user-attachments/files/21032516/log.log)

[log.log](https://github.com/user-attachments/files/21032518/log.log)

[configuration.txt](https://github.com/user-attachments/files/21032524/configuration.txt)

francisp2, 2025-07-03:
if 'Permit join' is enabled, this is what you see : 

[image]

Nerivec, 2025-07-03:
Yes `permit_join` in config was removed in 2.0.0 as this posed security risks (as well as issues with some firmware) due to being enabled all the time.