Koenkk/zigbee2mqtt #27891: EcoDim Eco-Dim.07 wrongly marked as “NOT supported”

jwjdaamen, 2025-07-03:
### What happened?

After pairing my dimmer (EcoDim.07 Zigbee Basic), Zigbee2MQTT successfully completes interview and configuration, but still logs:

`Device '0xfc4d6afffe36bd46' with Zigbee model 'eco-dim07-zigbee' and manufacturer name 'EcoDim BV' is NOT supported`

This is unexpected because the device is [listed as supported on your website](https://www.zigbee2mqtt.io/devices/Eco-Dim.07_Eco-Dim.10.html).

No external converter or custom configuration is used. It appears that Z2M is assigning an “automatically generated definition” instead of matching it to the proper one.

### What did you expect to happen?

I expected the device to be recognized as supported and not flagged as “NOT supported”.

### How to reproduce it (minimal and precise)

1. Put EcoDim.07 in pairing mode
2. Click on enable pairing in Z2M
3. Wait for pairing to complete
4. Error shows up

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

0x26780700

### Adapter

ConBee II

### Setup

	•	Add-on in Home Assistant OS
	•	Version: Home Assistant Core 2025.6.1
	•	Supervisor: 2025.06.2
	•	OS: 15.2
	•	Device: Raspberry Pi 4 (rpi4-64)

**Extra**
zigbee_herdsman_converters_version: 23.69.0
zigbee_herdsman_version:  4.3.1

### Debug log

[debug.log](https://github.com/user-attachments/files/21034172/debug.log)

Nerivec, 2025-07-03:
Looks like you have a variant (or firmware version) that's not currently supported.
Can you post the line entry in `database.db` for this device so someone can add the proper fingerprint?

jwjdaamen, 2025-07-03:
`{
  "id": 6,
  "type": "Router",
  "ieeeAddr": "0xfc4d6afffe36bd46",
  "nwkAddr": 11251,
  "manufId": 4714,
  "manufName": "EcoDim BV",
  "powerSource": "Mains (single phase)",
  "modelId": "eco-dim07-zigbee",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 257,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        2821,
        4096
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "eco-dim07-zigbee",
            "manufacturerName": "EcoDim BV",
            "powerSource": 46,
            "zclVersion": 8,
            "appVersion": 0,
            "stackVersion": 0,
            "hwVersion": 0,
            "dateCode": "20230401",
            "swBuildId": "1.01"
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 0,
  "stackVersion": 0,
  "hwVersion": 0,
  "dateCode": "20230401",
  "swBuildId": "1.01",
  "zclVersion": 8,
  "interviewCompleted": true,
  "interviewState": "SUCCESSFUL",
  "meta": {
    "configured": 332242049
  },
  "lastSeen": 1751538169886
}`

Koenkk, 2025-07-04:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

jwjdaamen, 2025-07-04:
@Koenkk I noticed that zigbee-herdsman was updated to 4.3.2, but zigbee-herdsman-converters is still at 23.69.0. Is that correct?

Koenkk, 2025-07-04:
Should be there in a few mins now!

jwjdaamen, 2025-07-05:
@Koenkk Works like a charm, it gets detected now. Thank you very much!