Koenkk/zigbee2mqtt #27902: 2.5.1-1 : impossible to permit join

phklrz, 2025-07-04:
### What happened?

After upgrading to 2.5.1-1, when trying to toggle permit join the following error appears:
`z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Failed APS-DATA.request with confirm status: unknown (0xa6)'`

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

0x26580700

### Adapter

Conbee II

### Setup

Add-on on Home Assistant on home assistant green

manup, 2025-07-04:
I can't re-create the error also with old firmware version 0x26580700 it works here when pressing the "All" join button.

```
[2025-07-04 17:38:14] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"we35h-1"}'
[2025-07-04 17:38:14] info: 	z2m: Zigbee: allowing new devices to join.
[2025-07-04 17:38:14] debug: 	zh:deconz:driver: Send write parameter request from queue. seq: 237 parameter: STK_PERMIT_JOIN
[2025-07-04 17:38:14] debug: 	zh:deconz:frameparser: Write parameter response parameter: STK_PERMIT_JOIN, status: Success
[2025-07-04 17:38:14] debug: 	zh:deconz: Permit joining on coordinator for 254 sec.
[2025-07-04 17:38:14] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0xfffc EP:0 seq: 238 requestId: 1
[2025-07-04 17:38:14] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 1 status: Success
[2025-07-04 17:38:15] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 246
[2025-07-04 17:38:15] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0xfffc APS request id: 1 confirm status: Success 0x00
[2025-07-04 17:38:15] debug: 	zh:deconz: zclFrame to all - 242
[2025-07-04 17:38:15] debug: 	zh:deconz: sendZclFrameToAll - message send
[2025-07-04 17:38:15] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0xfffd EP:242 seq: 247 requestId: 2
[2025-07-04 17:38:15] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 2 status: Success
[2025-07-04 17:38:16] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 1
[2025-07-04 17:38:16] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0xfffd APS request id: 2 confirm status: Success 0x00
[2025-07-04 17:38:16] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"we35h-1"}'
```

manup, 2025-07-04:
Unicast Permit Join via specific router seems to works as well in my setup:

```
[2025-07-04 17:42:21] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"0x7cb03eaa00add426","time":254,"transaction":"we35h-5"}'
[2025-07-04 17:42:21] info: 	z2m: Zigbee: allowing new devices to join via 0x7cb03eaa00add426.
[2025-07-04 17:42:21] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x5674 EP:0 seq: 192 requestId: 10
[2025-07-04 17:42:21] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 10 status: Success
[2025-07-04 17:42:21] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 194
[2025-07-04 17:42:21] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x5674 APS request id: 10 confirm status: Success 0x00
[2025-07-04 17:42:21] debug: 	zh:deconz:driver: Request APS-DATA.indication seq: 195
[2025-07-04 17:42:21] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 195 srcAddr: 0x5674 destAddr: 0x0 profile id: 0x0000 cluster id: 0x8036 lqi: 255
[2025-07-04 17:42:21] debug: 	zh:controller: Received ZDO response: clusterId=PERMIT_JOINING_RESPONSE, status=SUCCESS, payload=undefined
[2025-07-04 17:42:21] debug: 	zh:deconz: ZCL request sent with transactionSequenceNumber.: 6
[2025-07-04 17:42:21] debug: 	zh:deconz: command.response: undefined, zcl.disableDefaultResponse: true, z2m.disableResponse: false, request.timeout: 10000
[2025-07-04 17:42:21] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x5674 EP:242 seq: 196 requestId: 11
[2025-07-04 17:42:21] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 11 status: Success
[2025-07-04 17:42:21] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 198
[2025-07-04 17:42:21] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x5674 APS request id: 11 confirm status: Success 0x00
[2025-07-04 17:42:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"0x7cb03eaa00add426","time":254},"status":"ok","transaction":"we35h-5"}'
```

jurgverhoeven, 2025-07-05:
I seem to be having the same issue after updating today.
`Error 2025-07-05 16:11:58z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Failed APS-DATA.request with confirm status: unknown (0xa6)'`
I'm using the same adapter version.

Lachfrosch, 2025-07-05:
I had to do a fresh setup of home assistant today because I moved. Have the same error popping up, setup worked perfectly before.

why83key, 2025-07-06:
Same here: with 2.4.0 adding new devices worked well. Then with 2.5.0:
`info: 	z2m: Zigbee: allowing new devices to join.`
`error: 	z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Failed APS-DATA.request with confirm status: unknown (0xa6)'`

Already known devices can be controlled, however.

Coordinator firmware version: 0x266b0700 (ConBee2)

Rendeik, 2025-07-06:
I own a first series conbee, after updating to version 2.5.1 it no longer communicates with the devices and I can't even pair them again, I had to revert it to the previous version.

manup, 2025-07-06:
Uff that was tricky, finally managed to re-create the 0xA6 error:

```
z2m: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Failed APS-DATA.request with confirm status: unknown (0xa6)'
[2025-07-06 18:12:01] debug: 	z2m: Error: Failed APS-DATA.request with confirm status: unknown (0xa6)
    at parseApsConfirmResponse (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:203:20)
    at parseFrame (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:643:32)
    at processFrame (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:589:53)
    at Driver.<anonymous> (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:79:58)
    at Driver.emit (node:events:507:28)
    at Driver.onParsed (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:1479:22)
    at Parser.emit (node:events:507:28)
    at Parser.onMessage (/home/mpi/src/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/home/mpi/src/zigbee-herdsman/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:189:18)
    at slip.Decoder.p.decode (/home/mpi/src/zigbee-herdsman/node_modules/.pnpm/slip@1.0.2/node_modules/slip/src/slip.js:143:32)
```

Working a fix.

Lachfrosch, 2025-07-06:
Thanks for looking into this, looking forward to the fix 👍

closed 2025-07-06 by Koenkk

phklrz, 2025-07-07:
Thank you. Installed via zigbee2mqtt edge add_on in home assistant. Works without problem.