Koenkk/zigbee2mqtt #27904: Home Assistant ID not updating regardless of whether the corresponding checkbox is ticked or not

ruipin, 2025-07-04:
### What happened?

When I rename a device and check the "Update Home Assistant entity ID" checkbox, I'd expect the Home Assistant entity ID to be updated.

### What did you expect to happen?

For example, if a device is called "Foo" and has HASS entity ID "sensor.foo", and I rename it to "Bar" with that checkbox ticked, I expect the sensor to be renamed to "sensor.bar".

### How to reproduce it (minimal and precise)

Rename device with "Update Home Assistant ID" checkbox checked, and confirm whether the Home Assistant entity ID has updated or not.

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Docker container

ruipin, 2025-07-04:
I've created a HASS ticket https://github.com/home-assistant/core/issues/148172

GigiPompieru, 2025-07-05:
> I've hit this as well, on `2.5.1`. On HomeAssistant's side I'm seeing:
> 
> ```
> 2025-07-04 14:30:27.462 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.state }}'
> 2025-07-04 14:30:27.462 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.state }})
> 2025-07-04 14:30:27.462 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'value_json' is undefined when rendering '{{ value_json.power_on_behavior }}'
> 2025-07-04 14:30:27.462 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'value_json' is undefined (value: , template: {{ value_json.power_on_behavior }})
> 2025-07-04 14:30:27.463 WARNING (MainThread) [homeassistant.components.mqtt.update] Unable to process payload '' for topic zigbee2mqtt/0x04cd15fffe6a33ec, with value template 'Template<template=({"latest_version":"{{ value_json['update']['latest_version'] }}","installed_version":"{{ value_json['update']['installed_version'] }}","update_percentage":{{ value_json['update'].get('progress', 'null') }},"in_progress":{{ (value_json['update']['state'] == 'updating')|lower }}}) renders=1>'
> ```
> 
> This setup was working flawlessly until I tried to pair something today so it's something recent.
> 
> Edit: I tried [@ruipin](https://github.com/ruipin) 's downgrade to 2025.6 and confirmed it fixed it for me. Weirdly the console errors are still present, but it does in fact work.

does this mean that in 2025.7 you cannot add zigbee devices?

ruipin, 2025-07-05:
You can, but you'll be stuck with the `sensor.0xfoobar_temperature` etc entity names based on the hardware ID.

GigiPompieru, 2025-07-05:
> You can, but you'll be stuck with the `sensor.0xfoobar_temperature` etc entity names based on the hardware ID no matter what you do, unless you downgrade to 2025.6 and rename the device in Z2M again.

can't you just rename from HA as a workaround?

ruipin, 2025-07-05:
Yes, but assuming you want them to be consistent you'll have to rename every individual sensor, switch, etc which takes a while.

ruipin, 2025-07-05:
After doing it to 2 new devices I added, I gave up and just downgraded home assistant to add 2 other devices. It takes less time to downgrade than to do the renaming by hand 😅 

But yes, it is not a hard stop - and it is not like anything stops you from just using the hardware ID-based entities.

t-b-c-22, 2025-07-14:
Same issue here with 2025.7.0 and Z2M 2.5.1-1 and:

- the device name does change
- the sensor display name does change
- the entity id **does not** change

Koenkk, 2025-07-14:
This is a HA 2025.7 bug which has been fixed in https://github.com/home-assistant/core/pull/148484

closed 2025-07-14 by Koenkk