Koenkk/zigbee2mqtt #27912: Elivco LSPA9 appears permanently offline

Developer-Simon, 2025-07-05:
### What happened?

With the latest update, the [Elivco LSPA9](https://www.zigbee2mqtt.io/devices/LSPA9.html) Smart plugs are appearing permanently offline. I have 5 of them and all are having the same issue. I tried once to remove one and let it rejoin, but on attempt they trigger a reaction of Z2M but won't appear in the list at all.

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

0x26510900

### Adapter

ConBee3

### Setup

Home Assistant OS on Raspberry Pi 5

Developer-Simon, 2025-07-08:
I have added the debug log to the description. Surprisingly, the sockets still respond to commands from the frontend. But in HA they are offline and therefore not controllable.

Koenkk, 2025-07-08:
Could you check if this issue is fixed in the latest dev branch? (contains quite some Conbee fixes)

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Developer-Simon, 2025-07-13:
Sorry for the late response, I've tested with the 2.5.1-dev (Edge version on HA), unfortunately its still happening.

[image]

```
[2025-07-13 11:25:41] frontend:api: Sending {"topic":"0x70b3d52b600762c5/set","payload":{"state":"OFF"}}
[2025-07-13 11:25:43] frontend:api: Sending {"topic":"0x70b3d52b600762c5/set","payload":{"state":"ON"}}
[2025-07-13 11:25:52] z2m: Publish 'set' 'state' to 'Socket Office' failed: 'Error: ZCL command 0x70b3d52b600762c5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (No ZCL response received for (8) Error: waiting for response TIMEOUT)'
[2025-07-13 11:25:54] z2m: Publish 'set' 'state' to 'Socket Office' failed: 'Error: ZCL command 0x70b3d52b600762c5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (No ZCL response received for (9) Error: waiting for response TIMEOUT)'
```

Koenkk, 2025-07-13:
Could you try updating your Conbee to the latest version?