Koenkk/zigbee2mqtt #27915: Aqara Valve controller T1 reports 50% battery with brand new batteries

bourquep, 2025-07-06:
### What happened?

My Aqara Valve controller T1 is reporting battery at 50% even with brand new batteries:

[image]

### What did you expect to happen?

I'd expect to see 100% battery with brand new batteries.

### Zigbee2MQTT version

2.3.0

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

I installed z2m using the [Proxmox Community Script](https://community-scripts.github.io/ProxmoxVE/scripts?id=zigbee2mqtt) in a Proxmox LXC running on a Beelink Mini 12 Pro mini-PC.

```
Linux zigbee2mqtt 6.8.12-10-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-10 (2025-04-18T07:39Z) x86_64 GNU/Linux
Debian GNU/Linux 12
```

Koenkk, 2025-07-06:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-07-06 by Koenkk

Great-Chart, 2025-08-06:
I note that this doesn't appear to have fixed things for my device which still reports as 50%.
@bourquep - has this resolved things for yourself?

I note that the code change is basically don't divide by 2 and perhaps it assumes this unit follows other precedent whereas it might need to be multiply by 2?
https://github.com/Koenkk/zigbee-herdsman-converters/commit/e64eab876f9265630409c2674b79032cc669177d

Database entry for mine:-

{"id":169,"type":"EndDevice","ieeeAddr":"0x54ef441000de5333","nwkAddr":34083,"manufId":4447,"manufName":"Aqara","powerSource":"Battery","modelId":"lumi.valve.agl001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,3,0,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.valve.agl001","appVersion":53,"manufacturerName":"Aqara","powerSource":3,"zclVersion":3,"stackVersion":27,"hwVersion":1,"dateCode":"20240920"}},"manuSpecificLumi":{"attributes":{"12":{"type":"Buffer","data":[0,0,0,0,0,0,0,0]},"22":"351FAE87D47C6A3E4DC3A3E94B3D919627E5967AE8B445DA10F3B979E6CC42E6","223":{"type":"Buffer","data":[3,35,255,255,0,0,4,33,64,17,5,33,147,0,41,35,3,0,0,0,42,35,2,0,0,0,40,35,64,0,0,0,43,33,127,91,44,41,202,255,45,33,0,0]},"247":{"type":"Buffer","data":[3,40,27,5,33,19,0,9,33,0,0,10,33,89,174,12,32,10,13,35,53,0,0,0,17,35,1,0,0,0,110,32,1,111,32,100,112,32,0,113,65,8,105,213,37,48,0,0,0,0,121,32,0,154,32,0]}}},"genOnOff":{"attributes":{"245":126165760,"onOff":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1862","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1862","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":53,"stackVersion":27,"hwVersion":1,"dateCode":"20240920","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1754473704037}

bourquep, 2025-08-06:
Indeed, still showing 50% here as well.

Koenkk, 2025-08-10:
@Great-Chart is this with z2m 2.6.0?