Koenkk/zigbee2mqtt #27920: zigbee2mqtt Not starting

NMNige01, 2025-07-07:
### What happened?

Hi,

Up until the weekend zigbee2mqtt was running fine using a ZLB-06 as the coordinator.  Now if it does Strat it will not stay running for long.  Log pasted below, the IP address for the ZLB is correct as I can navigate to it using a browser.

[10:49:19] INFO: Preparing to start...
[10:49:19] INFO: Socat not enabled
[10:49:19] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 10:49:44] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 10:49:44] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 10:49:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 10:49:44] error: 	z2m: Exiting...
[2025-07-07 10:49:44] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:71:27)
    at Controller.start (/app/lib/controller.ts:102:13)
    at start (/app/index.js:149:5)
[10:50:43] INFO: Preparing to start...
[10:50:43] INFO: Socat not enabled
[10:50:44] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 10:51:09] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 10:51:09] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 10:51:09] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 10:51:09] error: 	z2m: Exiting...
[2025-07-07 10:51:09] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.1.0/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:71:27)
    at Controller.start (/app/lib/controller.ts:102:13)
    at start (/app/index.js:149:5)
[10:53:58] INFO: Preparing to start...
[10:53:58] INFO: Socat not enabled
[10:53:59] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 10:54:24] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 10:54:24] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 10:54:24] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 10:54:24] error: 	z2m: Exiting...
[2025-07-07 10:54:24] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:68:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
[10:54:53] INFO: Preparing to start...
[10:54:53] INFO: Socat not enabled
[10:54:54] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 10:55:19] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 10:55:19] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 10:55:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 10:55:19] error: 	z2m: Exiting...
[2025-07-07 10:55:19] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:68:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
[10:55:59] INFO: Preparing to start...
[10:55:59] INFO: Socat not enabled
[10:56:00] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 10:56:25] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 10:56:25] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 10:56:25] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 10:56:25] error: 	z2m: Exiting...
[2025-07-07 10:56:27] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:68:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
[11:03:47] INFO: Preparing to start...
[11:03:47] INFO: Socat not enabled
[11:03:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-07-07 11:03:55] error: 	zh:zstack:znp: Socket error Error: read ECONNRESET
[2025-07-07 11:04:01] error: 	z2m: Error while starting zigbee-herdsman
[2025-07-07 11:04:01] error: 	z2m: Failed to start zigbee-herdsman
[2025-07-07 11:04:01] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-07 11:04:01] error: 	z2m: Exiting...
[2025-07-07 11:04:01] error: 	z2m: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:68:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

v2.9.4

### Adapter

ZLB-06

### Setup

Running on a HA Green

francisp2, 2025-07-07:
Power off the SLZB-06, wait a minute, power it back on

NMNige01, 2025-07-08:
> Anything new on your network that could cause issues? Looks like the connection between Z2M and the adapter cuts-off repeatedly.
> Did you try downgrading the core firmware? I've seen troubles from 2.9.x in other issues.

Hi,Nothing new. The network for a few weeks, all running fine.  I had noticed a battery powered button had dropped off a few days earlier, before I could investigate the whole thing goes down!I’ve not tried downgrading the SLZB’s core firmware yet, 2.8.7 is the previous one showing up, have you seen any issues with that?