Koenkk/zigbee2mqtt #27922: Innr OFL 142 C, Power-on behavior - Unsupported Attribute

wills106, 2025-07-07:
### What happened?

Currently when power is lost to the Innr [OFL 142 C](https://www.zigbee2mqtt.io/devices/OFL_142_C.html#innr-ofl%2520142%2520c) lights and restored the lights automatically turn on.
I have tried to change the Power-on behavior, but when selecting a mode ie "off" or "previous" I get the error Unsupported Attribute

Power-on behaviour works on other devices such as my Hue [9290022890](https://www.zigbee2mqtt.io/devices/9290022890.html#philips-9290022890) and 
9290022891

### What did you expect to happen?

Power-on behavior to change to the mode selected

### How to reproduce it (minimal and precise)

Change Power-on behavior on [OFL 142 C](https://www.zigbee2mqtt.io/devices/OFL_142_C.html#innr-ofl%2520142%2520c) with Firmware build date 20221010 & Firmware ID 3.0.1

You will get the following error:

z2m: Publish 'set' 'power_on_behavior' to 'Canopy Light 4m' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

7.4.5 [GA]

### Adapter

EmberZNet - SONOFF Universal Zigbee 3.0 USB Dongle Plus ZBDongle E

### Setup

Unraid 7.1.2, Running zigbee2mqtt as Docker container through Apps Template

Linux Tower 6.12.24-Unraid #1 SMP PREEMPT_DYNAMIC Sat May  3 00:12:52 PDT 2025 x86_64 Intel(R) Core(TM) i7-9700T CPU @ 2.00GHz GenuineIntel GNU/Linux

### Debug log

[2025-07-07 12:56:35] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Canopy Light 4m/set' with data '{"power_on_behavior":"previous"}'
[2025-07-07 12:56:35] debug: 	z2m: Publishing 'set' 'power_on_behavior' to 'Canopy Light 4m'
[2025-07-07 12:56:35] debug: 	zh:controller:endpoint: ZCL command 0x00158d00091a29bd/1 genOnOff.write({"startUpOnOff":255}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-07-07 12:56:35] debug: 	zh:ember: ~~~> [ZCL to=0x00158d00091a29bd:22422 apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":43,"commandIdentifier":2}]
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=129 Len=28]
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=4](ackRx=1)
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=4](frmRx=4) Added to rxQueue
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=129 Len=7]
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=9 messageTag=103 status=OK]
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=5](frmRx=5) Added to rxQueue
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=129 Len=22]
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: ezspMessageSentHandler: status=OK type=DIRECT indexOrDestination=22422 apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":9} messageTag=103
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=6](frmRx=6) Added to rxQueue
[2025-07-07 12:56:35] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=2)
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=129 Len=31]
[2025-07-07 12:56:35] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":115} packetInfo:{"senderShortId":22422,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":184,"lastHopRssi":-54,"lastHopTimestamp":0} messageContents=182b04860340
[2025-07-07 12:56:35] debug: 	zh:controller: Received payload: clusterID=6, address=22422, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=184, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":43,"commandIdentifier":4},"payload":[{"status":134,"attrId":16387}],"command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}}
[2025-07-07 12:56:35] debug: 	zh:controller:endpoint: Error: ZCL command 0x00158d00091a29bd/1 genOnOff.write({"startUpOnOff":255}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:363:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:987:26)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at runNextTicks (node:internal/process/task_queues:69:3)
    at processImmediate (node:internal/timers:453:9)
    at Endpoint.write (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:406:9)
    at Object.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:314:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:245:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17)
[2025-07-07 12:56:35] error: 	z2m: Publish 'set' 'power_on_behavior' to 'Canopy Light 4m' failed: 'Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour'
[2025-07-07 12:56:35] debug: 	z2m: Error: Got `UNSUPPORTED_ATTRIBUTE` error, device does not support power on behaviour
    at Object.convertSet (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:317:23)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at runNextTicks (node:internal/process/task_queues:69:3)
    at processImmediate (node:internal/timers:453:9)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:245:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17)

wills106, 2025-09-06:
Not stale, still got the same issue on [2.6.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.1) commit: [e9889e5d20e8911550d5231bc0349567cc2bb9ce](https://github.com/Koenkk/zigbee2mqtt/commit/e9889e5d20e8911550d5231bc0349567cc2bb9ce)

Koenkk, 2025-09-06:
`UNSUPPORTED_ATTRBIUTE` means the device doesn't support it, I disabled it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-09-06 by Koenkk

wills106, 2025-09-06:
That's a shame. I don't have an Innr hub, so can't see if there is a Firmware update to fix the issue.
It's annoying the light turns back on after a power failure.