Koenkk/zigbee2mqtt #27926: [Feature request]: add genAnalogInput/(genAnalogOutput) clusters to auto generator

luar123, 2025-07-07:
### Is your feature request related to a problem? Please describe

We are trying to add zigbee support for all esphome sensors. Some could use special clusters (like temperature) others could use this generic analog input cluster (not sure if there is also a usecase for analog output, but it would be very similar). Additionally the description attribute is very useful.
Compared to the binary clusters `Application Type` or `Engineering Units` clusters need to be evaluated to get the correct unit/type.

### Describe the solution you'd like

generators similar to https://github.com/Koenkk/zigbee-herdsman-converters/pull/7913
with some logic to get the unit similar to the ZHA implementation: https://github.com/zigpy/zha/pull/197/files#diff-d003cdf3ae320a84576be48fc6cd8a446c5aecb7949a038d2601b72d6be35c72

### Describe alternatives you've considered

Custom clusters could be an alternative, but wouldn't be easier to auto-detect I guess.

### Additional context

.

luar123, 2025-09-06:
Not stale

Koenkk, 2025-09-06:
@luar123 would you mind take a stab at this and make a PR?

luar123, 2025-09-07:
If no one goes for it I will give it a try once my esphome ZigBee implementation is more usable, as I would prefer to use z2m over zha. But I have 0 typescript experience.

Also @tomaszduda23 might look into it.

tomaszduda23, 2025-09-13:
presentValue and description implemented by https://github.com/Koenkk/zigbee-herdsman-converters/pull/10004.

luar123, 2025-09-13:
@tomaszduda23 Awesome, thank you!

At some point it would be great to also map the units/application type.

tomaszduda23, 2025-09-14:
> [@tomaszduda23](https://github.com/tomaszduda23) Awesome, thank you!
> 
> At some point it would be great to also map the units/application type.

how would you like to do that? https://github.com/Koenkk/zigbee-herdsman-converters/blob/ccb67440ab61bee06b1374d5c0281f1d53a50fbd/src/lib/modernExtend.ts#L2572 there is `unit`. I cannot see any easy mapping inside https://csa-iot.org/wp-content/uploads/2022/01/07-5123-08-Zigbee-Cluster-Library-1.pdf

tomaszduda23, 2025-09-14:
@luar123 there is also `label` which seems like type of device (application type). Still not sure how to map it.

luar123, 2025-09-14:
I think this would need a big dictionary to map each value of the application type or unit attribute to the unit string.