Koenkk/zigbee2mqtt #27935: Failed to start 'Bridge' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')

ronsmits, 2025-07-08:
### What happened?

Brand new install, Conbee II usb stick, always worked on a pi, but the pi broke. So installing a fresh zigbee2mqtt on a PC. Stick is recognized. but it gives the `Failed to start 'Bridge' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')` I tried the stable version and the dev version. Here is the log file: https://pastebin.com/wX7h6m7N

### What did you expect to happen?

I expected the zigbee2mqtt to start :)

### Zigbee2MQTT version

2.5.1-dev (commit #93c1824b)

### Adapter firmware version

latest version from fiurmware update deCONZ_ConBeeII_0x26780700.bin.GCF

### Adapter

USB Conbee II

### Setup

Linux hawk 6.1.0-37-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64 GNU/Linux
Debian GNU/Linux 12

### Debug log

[log.log](https://github.com/user-attachments/files/21130411/log.log)

manup, 2025-07-08:
Not sure if this is is an error in the adapter. The logs show that the IeeeAddress gets successful read from the firmware and is also queried by z2m from the adapter in the `getCoordinatorIEEE()` function

```
[2025-07-08 21:35:36] debug: 	zh:deconz:frameparser: Received read parameter response for MAC_ADDRESS, seq: 4, status: 0, parameter: 0x00212effff0759b2
```

```
[2025-07-08 21:35:37] debug: 	zh:deconz: -------- z2m:getCoordinatorIEEE() ----------------
[2025-07-08 21:35:37] debug: 	zh:controller:device: Request Queue (0x00212effff0759b2): default expiration timeout set to 0
```

So question is what causes the following 🤔 

```
[2025-07-08 21:35:37] error: 	z2m: Failed to start 'Bridge' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at Bridge.publishInfo (/opt/zigbee2mqtt/lib/extension/bridge.ts:707:69)
    at Bridge.start (/opt/zigbee2mqtt/lib/extension/bridge.ts:186:9)
    at Controller.startExtension (/opt/zigbee2mqtt/lib/controller.ts:270:13)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:148:13)
    at start (/opt/zigbee2mqtt/index.js:149:5))
```

Nerivec, 2025-07-09:
Looks like it's trying to identify devices after startup, so, some sync issue between the network and the database.db seems probable. Maybe the Conbee has a saved network, but the database.db is empty (just an incomplete coordinator entry)?

I'm seeing some weird things in the MQTT messages too:
- "version":"1.28.0"
- `Received MQTT message on 'zigbee2mqtt/bridge/config'`

Yet it is indeed 2.5.1-dev on startup....!?

mbrossi, 2025-07-10:
I am facing the same issue with the latest version, using a Conbee III.

The issue seems to be related with `zigbee-herdsman` in version **4.2.0** and higher. 

I changed the version in package.json to **4.1.2**, deleted the coordinator in `data/database.db` and did the onboarding step once again. It went through and the coordinator record now also show endpoints.

mbrossi, 2025-07-10:
I guess I found the issue on my end:

My ConBee III was still using the initial firmware `26450900` (November 2023). After updating the firmware to `26550900` (April 2025) it works like charm with `zigbee-herdsman` in version **4.3.1**.

If you have the same issue, you can either update the firmware via the Phoscon app, or using the [GCFFlasher](https://github.com/dresden-elektronik/gcfflasher). The firmware can be downloaded at [https://deconz.dresden-elektronik.de/deconz-firmware/](https://deconz.dresden-elektronik.de/deconz-firmware/).

Koenkk, 2025-07-11:
@manup given that there are quite some issues with the older fws, maybe it makes sense to log a warning on startup when using an older fw?

Nerivec, 2025-07-11:
The error itself means there is no endpoint for the coordinator saved in `database.db`.

But the weirdness with MQTT data suggests something is seriously corrupted somewhere (like Z2M is a mix of 2 versions).
Try clearing everything and rebuilding.

```sh
rm -rf node_modules
pnpm i --frozen-lockfile
pnpm run prepack
```

ronsmits, 2025-07-17:
so it took me some time but I made a complete clean install and every works fine which makes me happy and sad as I am trying to restore an old crashed install, I hoped restoring the data directory would have been enough

closed 2025-07-22 by ronsmits