Koenkk/zigbee2mqtt #27950: [Bug] ThirdReality 3RSM0147Z - soil_moisture always null (firmware v38)

mauriziosavio, 2025-07-09:
### What happened?

I've installed multiple Third Reality 3RSM0147Z soil moisture sensors (firmware v38).  
They expose the `soil_moisture` property in MQTT, but its value is always `null`, even after 24h in wet soil.

### Observations
- The device joins Zigbee2MQTT correctly and reports temperature, humidity, battery.
- Cluster `msSoilMoisture` is missing (read command fails with `UNSUPPORTED_ATTRIBUTE`).
- Tried manually binding `3rSoilSpecialCluster` and configuring reporting (no success).
- Device doesn’t respond to `/get` commands for `soil_moisture`.
- Tried with 6 different units – same result.

### What I suspect
- Either the sensor firmware is not sending the value
- Or Zigbee2MQTT’s converter might expect it on a wrong cluster
- Or it uses a custom attribute (like analog_input) that needs decoding

Could you please help check if the converter should be updated, or suggest a test?

Thanks!

### What did you expect to happen?

I expected to receive valid numeric values in `soil_moisture`, as documented on the device support page.

### How to reproduce it (minimal and precise)

![Image](https://github.com/user-attachments/assets/7

[image]

7dc4965-7530-4ebe-9dc8-375f3f9b8718)

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

7.4.4 [GA]

### Adapter

ZBDongle-E

### Setup

Home Assistant OS add-on su Raspberry Pi 4

agherzan, 2025-07-10:
Same issue here. I suspect that the data is not provided as "humidity".

sajikur, 2025-07-11:
same here with the last z2m update
with the previous firmware only the three values ​​were active:
- temperature
- humidity
- battery

maurizioandreotti60, 2025-07-11:
I see that the issue has been reported.

agherzan, 2025-07-12:
I have tested these with Home assistant for the past 2 weeks. They are extremely inaccurate when having an ecowitt system to compare against. This issue is minor compared to the fact that these sensors are useless.

Maurizion, 2025-07-12:
> I have tested these with Home assistant for the past 2 weeks. They are extremely inaccurate when having an ecowitt system to compare against. This issue is minor compared to the fact that these sensors are useless.

How could you test them and compare them with the ecowitt sensors (which I also have and I confirm that they are excellent) if it does not correctly report soil moisture in HA?

agherzan, 2025-07-12:
> How could you test them and compare them with the ecowitt sensors (which I also have and I confirm that they are excellent) if it does not correctly report soil moisture in HA?

I have used the humidity entity.

agherzan, 2025-07-12:
I don't think this device has an air humidity sensor. And you can see that because if you take the probe out of the ground, it goes to 0%.

Maurizion, 2025-07-12:
> I don't think this device has an air humidity sensor. And you can see that because if you take the probe out of the ground, it goes to 0%.

If humidity should be interpreted as soil humidity, then I have a 25% higher value compared to the Ecowtt sensors which is not realistic or is the indicator of a defective product

agherzan, 2025-07-12:
@Maurizion You can see screenshots in their official manual[image]

And yes, they seem to be that inaccurate. Loads of other posts say the same.

3reality-support, 2025-07-14:
Although soil moisture is always null, Humility can be used. In fact, the reality of Humidity is the soil moisture measured by sensors

abonforti, 2025-07-14:
> Yes, this is a bug that will be fixed in the next version

We understood, it’s been confirmed several times. What is still unclear and perhaps you can clarify is the release date of this firmware version.
Can you specify a day?

3reality-support, 2025-07-14:
> > Yes, this is a bug that will be fixed in the next version
> 
> [@3reality-support](https://github.com/3reality-support) we understood, it’s been confirmed several times. What is still unclear and perhaps you can clarify is the release date of this firmware version. Can you specify a day?

I can't be sure about this either, because only when all test cases pass can a new version be released. Our testers are already accelerating the progress

Marcoletto, 2025-07-15:
Guys, I pair this 3RSM0147Z device today and, unfortunately, I' m not able to se both Humidity and Moisture.
Temperature, battery and LQI are visible and they are working.

Following some pictures:

[image]
[image]
[image]

agherzan, 2025-07-21:
> > My solution: ecowitt. The consistency of measurements is of a completely different level.
> 
> But you need the Gateway (just for those sensors) and can not use a an existing infrastructure. And it seems to be limited to 8 sensors?
> 
> 
> @3reality-support Please speed up to fix the firmware. I can not believe that such a simple sensor needs so much time to implement the firmware. Or do you have problems with the hardware?

True. I put up with that overhead and cost for the value of a usable system.

3reality-support, 2025-07-24:
Yes, you can only use soil moisture and ignore humidity. It is expected that humidity will be removed in the next z2m version

> [Koenkk/zigbee-OTA#817](https://github.com/Koenkk/zigbee-OTA/pull/817) adds firmware v1.00.46 via OTA. I just updated and soil moisture now reports the same value as humidity: <img alt="image" width="358" height="204" src="https://private-user-images.githubusercontent.com/4833332/470048248-7e0f022b-57af-4312-a284-0792e9455d15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMzMjk3MDcsIm5iZiI6MTc1MzMyOTQwNywicGF0aCI6Ii80ODMzMzMyLzQ3MDA0ODI0OC03ZTBmMDIyYi01N2FmLTQzMTItYTI4NC0wNzkyZTk0NTVkMTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDcyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MjRUMDM1NjQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGQxYzQxMzc0NTAxYTU2YjVhMTc5Zjc5MWYyNGIyMWM2ZmUyMjcyMDQwMzZhMTkwNzA5MWI1OThmZDlhMTc4NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.IqZgGAbcWhxu5PU0q3PyOqvuYrt3yIz5cVVmYKcXxmU">

3reality-support, 2025-07-28:
We have informed the relevant personnel and will update within the next few days

> Thanks [@3reality-support](https://github.com/3reality-support), are there plans to add release notes to https://3reality.com/release-note/? Does the new firmware change anything besides adding soil moisture?

MSL-DA, 2025-07-28:
> We have informed the _relevant personnel_

LOL

chrisn997, 2025-09-25:
> [Koenkk/zigbee-OTA#817](https://github.com/Koenkk/zigbee-OTA/pull/817) adds firmware v1.00.46 via OTA. I just updated and soil moisture now reports the same value as humidity: <img alt="image" [...]

Same here.

sbraz, 2025-09-26:
@Koenkk you can close this. The firmware update addresses the issue.

closed 2025-09-26 by Koenkk

3reality-support, 2025-12-12:
> pdated the firmware to .46 and i get humidity measur

After upgrading the firmware, it is necessary to add the device again

KameDomotics, 2025-12-12:
> > pdated the firmware to .46 and i get humidity measur
> 
> After upgrading the firmware, it is necessary to add the device again

Hi
I haven't added the sensor again and it still works. But humidity sensor and soil moisture sensor have the same readings. Is that correct?

maurizioandreotti60, 2025-12-12:
> > pdated the firmware to .46 and i get humidity measur
> 
> After upgrading the firmware, it is necessary to add the device again

I have updated the fw but it was not necessary to repair the device. What it is wrong if I don't pair again the device?

3reality-support, 2025-12-12:
> > > > > pdated the firmware to .46 and i get humidity measur
> > > > 
> > > > 
> > > > After upgrading the firmware, it is necessary to add the device again
> > > 
> > > 
> > > Hi I haven't added the sensor again and it still works. But humidity sensor and soil moisture sensor have the same readings. Is that correct?
> > 
> > 
> > It is correct, what is actually measured is soil moisture
> 
> so why two different sensors

Because other HT platforms do not support soil moisture and can only display humidity, two types of humidity were used for compatibility