Koenkk/zigbee2mqtt #27956: [New device support]: 'TS0601', ['_TZE284_o9ofysmo',_TZE284_xc3vwx5a]

MasterFeige, 2025-07-10:
### Link

Unpublished

### Database entry

{"id":26,"type":"EndDevice","ieeeAddr":"0xa4c138825ad81fa0","nwkAddr":29554,"manufId":4417,"manufName":"_TZE284_o9ofysmo","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":30,"65503":"'K\u00010\u0011","65506":56,"65508":0,"appVersion":80,"modelId":"TS0601","manufacturerName":"_TZE284_o9ofysmo","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1752049467694}

### Zigbee2MQTT version

2.5.1

### Comments

Here are the definitions related to Tuya DP points:
[
{
"dpid": 3,
"identifier": "humidity",
"description": "Soil moisture",
"data_type": "Value",
"value_range": [0,100],
"unit": "%",
"remarks": ""
},
{
"dpid": 5,
"identifier": "temp_current",
"description": "Current ambient temperature",
"data_type": "Value",
"value_range": [-400,1200],
"unit": "°C",
"remarks": "Divide the value by 10"
},
{
"dpid": 14,
"identifier": "battery_state",
"description": "Current battery level",
"data_type": "Enum",
"enum_values": ["low", "middle", "high"],
"remarks": ""
},
{
"dpid": 101,
"identifier": "env_humidity",
"description": "Ambient humidity",
"data_type": "Value",
"value_range": [0,100],
"unit": "%RH",
"remarks": ""
},
{
"dpid": 102,
"identifier": "dimmer",
"description": "Ambient light intensity",
"data_type": "Value",
"value_range": [0,10000],
"unit": "Lux",
"remarks": ""
},
{
"dpid": 103,
"identifier": "adjust_humidity",
"description": "Soil moisture calibration",
"data_type": "Value",
"value_range": [-30,30],
"unit": "%RH",
"remarks": ""
},
{
"dpid": 104,
"identifier": "adjust_sample_time",
"description": "Sampling time calibration",
"data_type": "Value",
"value_range": [30,1200],
"unit": "s",
"remarks": ""
}
]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints={
  ...tuya.tz.datapoints,
  key:[
    'humidity',
    'temp_current',
    'battery_state',
    'env_humidity',
    'dimmer',
    'adjust_humidity',
    'adjust_sample_time',
  ]
}
const definition = {

    fingerprint:tuya.fingerprint('TS0601', ['_TZE284_o9ofysmo','_TZE284_xc3vwx5a']),
    model: 'ZS-301Z',
    vendor: 'arteco',
    description: 'Suitable for agriculture and home planting, this sensor is capable of detecting soil moisture, light intensity, and ambient temperature and humidity. It adopts low power consumption to ensure long-term operation of the device.',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.enum('battery_state',ea.STATE,['low','middle','high']).withDescription('low: 1-25%, middle: 26-50%, high: 51-100%'),
        e.numeric('humidity',ea.STATE).withUnit('%').withDescription('Soile Relative humidity').withValueMin(0).withValueMax(100),
        e.numeric('temp_current',ea.STATE).withUnit('°C').withDescription('Current temperature in the environment').withValueMin(-400).withValueMax(1200),
        e.numeric('env_humidity',ea.STATE).withUnit('%RH').withDescription('Relative humidity in the environment').withValueMin(0).withValueMax(100),
        e.numeric('dimmer',ea.STATE).withUnit('lux').withDescription('Dimmer value').withValueMin(0).withValueMax(10000),
        e.numeric('adjust_humidity',ea.STATE).withUnit('%RH').withDescription('Adjust humidity').withValueMin(-30).withValueMax(30),
        e.numeric('adjust_sample_time',ea.STATE).withUnit('s').withDescription('Adjust sample time').withValueMin(30).withValueMax(1200),
    ],
    meta: {
        tuyaDatapoints: [
          [3,'humidity',tuya.valueConverter.raw],
          [5,'temp_current',tuya.valueConverter.divideBy10],
          [14,'battery_state',tuya.valueConverterBasic.lookup({'low':tuya.enum(0),'middle':tuya.enum(1),'high':tuya.enum(2)})],
          [101,'env_humidity',tuya.valueConverter.raw],
          [102,'dimmer',tuya.valueConverter.raw],
          [103,'adjust_humidity',tuya.valueConverter.raw],
          [104,'adjust_sample_time',tuya.valueConverter.raw]
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

This is a new device, and I hope all new devices of this model can directly use the current conversion script.

Koenkk, 2025-07-11:
Could you give me a link to the pr?

MasterFeige, 2025-07-16:
Hi,Bro,this is pr link:https://www.aliexpress.com/item/1005009505575633.html

MasterFeige, 2025-08-12:
Hi there, buddy. I noticed there's a discrepancy between the integrated conversion script and the external conversion script I provided. In the code lines 7073 to 7109 of zigbee-herdsman-converters/src/devices/tuya.ts,
``` {
        fingerprint: tuya.fingerprint("TS0601", ["_TZE284_o9ofysmo", "_TZE284_xc3vwx5a"]),
        model: "ZS-301Z",
        vendor: "Arteco",
        description: "Soil moisture sensor",
        extend: [tuya.modernExtend.tuyaBase({dp: true, timeStart: "2000"})],
        exposes: [
            e.enum("battery_state", ea.STATE, ["low", "middle", "high"]).withDescription("low: 1-25%, middle: 26-50%, high: 51-100%"),
            e.temperature(),
            e.humidity(),
            e.illuminance(),
            e
                .numeric("humidity_calibration", ea.STATE_SET)
                .withUnit("%")
                .withDescription("Adjust humidity")
                .withValueMin(-30)
                .withValueMax(30)
                .withValueStep(1),
            e
                .numeric("report_interval", ea.STATE_SET)
                .withUnit("s")
                .withDescription("Report interval")
                .withValueMin(30)
                .withValueMax(1200)
                .withValueStep(30),
        ],
        meta: {
            tuyaDatapoints: [
                [3, "humidity", tuya.valueConverter.raw],
                [5, "temperature", tuya.valueConverter.divideBy10],
                [14, "battery_state", tuya.valueConverterBasic.lookup({low: tuya.enum(0), middle: tuya.enum(1), high: tuya.enum(2)})],
                [101, "humidity", tuya.valueConverter.raw],
                [102, "illuminance", tuya.valueConverter.raw],
                [103, "humidity_calibration", tuya.valueConverter.raw],
                [104, "report_interval", tuya.valueConverter.raw],
            ],
        },
    },```
I found that Tuya's DP points 3 and 101 use the same key =>"humidity".This causes only one value to be displayed on the HA panel. 
Actually, DP 3 should be accurately described as soil moisture, and DP 101 as ambient humidity. Could you please fix this? Thank you very much, and I look forward to your reply.