Koenkk/zigbee2mqtt #27957: [Wrong device]: ZG-101Z/D recognized as TS004F - _TZ3000_abrsvsou

AndTon35bis, 2025-07-10:
### Link

Product page : https://www.haozee.com/products/haozee-tuya-smart-zigbee-scene-switch-dimmer-button-knob-controller-for-home-automation-intelligent-diy-switch-support-home-assistant?_pos=2&_sid=4bf152674&_ss=r // Purchase page : https://www.domadoo.fr/fr/produits-zigbee/7907-haozee-bouton-variateur-rotatif-zigbee-tuya-smart-life-knob-button.html

### Model

ZG-101Z/D

### Description

Smart Knob Dimmer

### Vendor

Haozee (Tuya)

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZG-101Z-D.png

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c138cbfd2a533c","nwkAddr":40867,"manufId":4742,"manufName":"_TZ3000_abrsvsou","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_abrsvsou","powerSource":3,"zclVersion":3,"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"genOnOff":{"attributes":{"onOff":1,"tuyaOperationMode":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe406bffffe34a6c9","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe406bffffe34a6c9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe406bffffe34a6c9","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1752140691343}

### Notes

It's the same issue #25905

[image]

Koenkk, 2025-07-10:
is this with z2m 2.25.1?

AndTon35bis, 2025-07-10:
> is this with z2m 2.25.1?

Nope, with the lastest (2.5.1-1).

AndTon35bis, 2025-07-10:
> For some reason it doesn't seem to be detected as supported at all, can you provide a screenshot of the frontend -> settings -> about tab?

[image]

[image]

closed 2025-07-11 by Koenkk