Koenkk/zigbee2mqtt #27958: Z2M fail to start since update

JohnnyDarko23, 2025-07-10:
### What happened?

I cant fix this problem: 

TypeError: Cannot create property 'port' on string '/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_488e737465d8ed11bbea786162c613ac-if00-port0'
    at iterate (/app/lib/util/settings.ts:411:63)
    at iterate (/app/lib/util/settings.ts:424:21)
    at iterate (/app/lib/util/settings.ts:424:21)
    at applyEnvironmentVariables (/app/lib/util/settings.ts:430:5)
    at Object.write (/app/lib/util/settings.ts:250:5)
    at onboard (/app/lib/util/onboarding.ts:536:18)
    at start (/app/index.js:137:25)

can anybody help?

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

latest

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Raspberry Pi 5
rpi5-64

Koenkk, 2025-07-10:
can you provide your configuration.yaml file?

Koenkk, 2025-07-10:
Could you make a screenshot such that I can see the indentation?

JohnnyDarko23, 2025-07-10:
[image]

JohnnyDarko23, 2025-07-10:
[image]

Kyburg82, 2025-07-10:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: /dev/ttyUSB0
  adapter: zstack

Nerivec, 2025-07-10:
Set it to `serial: {}` similar to the line above (or if using the UI mode, just empty the `serial` field).
Can you also show the content of the actual `config/zigbee2mqtt/configuration.yaml` file (not the content of the add-on config page)?
You can use a file explorer add-on to see it, if you don't have anything already to read plain files on HA. Either will do:
- https://www.home-assistant.io/common-tasks/os/#installing-and-using-the-file-editor-add-on
- https://www.home-assistant.io/common-tasks/os/#installing-and-using-the-visual-studio-code-vsc-add-on

JohnnyDarko23, 2025-07-11:
[image]

JohnnyDarko23, 2025-07-11:
This is my configuration.yaml

Koenkk, 2025-07-11:
> This is my configuration.yaml

You need to put the "port" in the serial block (just like in the config you showed earlier)