Koenkk/zigbee2mqtt #27968: ZG-205ZL shows Exception while calling fromZigbee converter: Value '4' (and Value '5') is not allowed, expected one of 0,1,2,3} error

Aekung, 2025-07-11:
### What happened?

I have a problem with my newly added Presence Detector ZG-205ZL (24Ghz).

I have tried to look for the same problem reporting here and found 2 closed issues stating that it's already fixed. I somehow even tried the Edge version of Zigbee2Mqtt but still the error is still persisted.

Now sure what's the problem here. I'm on the latest version of the Home Assistant Addon and I believe I have updated my Sonoff Dongle P to the latest version already. Home Assistant is also in the latest version and I have also tried interviewing the device multiple times already.

The error is as below;

`2025-07-11 20:27:46z2m: Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of 0,1,2,3}`
`2025-07-11 20:28:05z2m: Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2,3}`

I understand from my research that it has something to do with the motion_state but I'm not sure what to do next or what I have done wrong since it should already be solved according to this

https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v21.17.0

### What did you expect to happen?

I expected these errors to already be fixed according to the link above.

### How to reproduce it (minimal and precise)

Just adding this device in the latest verion of Zigbee2Mqtt and it happens.

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

20210708

### Adapter

ZStack3x0

### Setup

Zigbee2Mqtt as an add-on in Home Assistant running in an intel core-i5 mini PC on Proxmox.

Aekung, 2025-07-12:
Hi,

Here is the debug log.

```
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'binary_sensor/0xa4c1384c0b7e0f3e/presence/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'sensor/0xa4c1384c0b7e0f3e/motion_state/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'sensor/0xa4c1384c0b7e0f3e/illuminance/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/fading_time/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/large_motion_detection_distance/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/large_motion_detection_sensitivity/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/small_motion_detection_distance/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/small_motion_detection_sensitivity/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/static_detection_distance/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/static_detection_sensitivity/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'select/0xa4c1384c0b7e0f3e/mode/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'select/0xa4c1384c0b7e0f3e/alarm_volume/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'number/0xa4c1384c0b7e0f3e/alarm_time/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'switch/0xa4c1384c0b7e0f3e/switch_light_mode/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'sensor/0xa4c1384c0b7e0f3e/linkquality/config', already discovered
[2025-07-12 18:55:35] debug: 	z2m: Skipping discovery of 'sensor/0xa4c1384c0b7e0f3e/last_seen/config', already discovered
[2025-07-12 18:55:38] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,239,50,118,1,1,0,87,0,47,114,141,0,0,10,9,72,2,0,72,11,4,0,1,4,50,118,29,165]
[2025-07-12 18:55:38] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,239,50,118,1,1,0,87,0,47,114,141,0,0,10,9,72,2,0,72,11,4,0,1,4,50,118,29,165]
[2025-07-12 18:55:38] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,50,118,1,1,0,87,0,47,114,141,0,0,10,9,72,2,0,72,11,4,0,1,4,50,118,29] - 165
[2025-07-12 18:55:38] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":30258,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":9269807,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,72,2,0,72,11,4,0,1,4]}}
[2025-07-12 18:55:38] debug: 	zh:controller: Received payload: clusterID=61184, address=30258, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=87, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":72,"commandIdentifier":2},"payload":{"seq":18432,"dpValues":[{"dp":11,"datatype":4,"data":{"type":"Buffer","data":[4]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2025-07-12 18:55:38] debug: 	zh:controller:endpoint: ZCL command 0xa4c1384c0b7e0f3e/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":72,"writeUndiv":false})
[2025-07-12 18:55:38] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c1384c0b7e0f3e:30258/1 (0,0,1)
[2025-07-12 18:55:38] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":30258,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":16,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,72,11,2,0]}}
[2025-07-12 18:55:38] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,50,118,1,1,0,239,16,0,30,5,16,72,11,2,0,219]
[2025-07-12 18:55:38] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-07-12 18:55:38] debug: 	z2m: Received Zigbee message from '0xa4c1384c0b7e0f3e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[4],"type":"Buffer"},"datatype":4,"dp":11}],"seq":18432}' from endpoint 1 with groupID 0
[2025-07-12 18:55:38] error: 	z2m: Exception while calling fromZigbee converter: Value '4' is not allowed, expected one of 0,1,2,3}
[2025-07-12 18:55:38] debug: 	z2m: Error: Value '4' is not allowed, expected one of 0,1,2,3
    at Object.from (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:544:27)
    at Object.convert (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.69.0/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1951:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:171:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:23)
    at EventEmitter.emit (node:events:530:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:174:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:128:27)
[... cut ...] rm_volume":"high","fading_time":10,"illuminance":1643,"large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-07-12T18:55:38+07:00","light_mode":"OFF","linkquality":87,"mode":"off","motion_state":"large","presence":true,"small_motion_detection_distance":6,"small_motion_detection_sensitivity":5,"static_detection_distance":4.5,"static_detection_sensitivity":5}'
```
Please let me know if I should provide anything else to settle this issue.

Thank you in advance.

Koenkk, 2025-07-12:
This is indeed related to the motion state.

> Should I pair it with Tuya Zigbee Hub to find out?

Can be done indeed, what you can try is pair it, trigger the same type of movement (such that it publishes either 4 or 5) and see what the app shows.

Aekung, 2025-07-12:
Based on the status after pairing the device to Tuya hub, I can see that it reports more than just 0 - 3 but I don't know what it is in particular order.

Now is the status that I could capture so far

None
Near
Far Away
Large Motion
Stillness

However, as the error mentioned Value 4 and 5, should it be that I miss 1 more status update? I have tried many things but these seem to be all the status it's reporting me thus far.

Aekung, 2025-07-12:
Could it be that the last status I couldn't get is Small Motion?

I see in the settings that there's a Large Motion parameter and Small Motion Parameter. Since now we do have a Large Motion, I would guess that the last one is the Small Motion but I still can't make it triggers in that state though.

Koenkk, 2025-07-14:
Based on this, any chance you could guess what value 4 and 5 mean?

Aekung, 2025-07-14:
> > * Currently the following values are mapped, we have to reverse engineer the other ones, do you have any clue where they should be mapped to?
> > 
> > ```
> >                     tuya.valueConverterBasic.lookup({
> >                         none: tuya.enum(0),
> >                         large: tuya.enum(1),
> >                         small: tuya.enum(2),
> >                         static: tuya.enum(3),
> >                     }),
> > ```
> 
> Can you instruct me on how to get find this for you? Should I pair it with Tuya Zigbee Hub to find out?
> 
> Thank you and looking forward to hearing from you.

Few days passed and I could see None, Large and Static states in the report.

If I have to guess, I think Small is not being triggered somehow but it's already there. But Near and Far Away are the two values that are missing.

I tried again and again until it leads to conclusion that 

4 is far away
5 is near

I think these are the missing values.