Koenkk/zigbee2mqtt #27970: ThirdReality 3RSM0147Z configuration loop

IgnacioPuchet, 2025-07-12:
### What happened?

I tried to add the third reality soil sensor 3RSM0147Z to the zigbee network but it fails in the configuration process with the message 

[image]

After that the device leaves the network and tries to rejoin.
On zigbee2mqtt version 2.4.0-1 configuration doesn't fail, but a few second later the device leaves the network and the loop continues.

### What did you expect to happen?

The device would stay in the network after beeing configured.

### How to reproduce it (minimal and precise)

Permit join in zigbee2mqtt, press reset button for 5 seconds in soil moisture sensor, wait until device joins network.

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

20221226

### Adapter

ZBDongle-P

### Setup

Add-on on Home Assistant OS on x86-64

### Debug log

[log.log](https://github.com/user-attachments/files/21192999/log.log)

Koenkk, 2025-07-12:
Could you provide the data/databasedb entry of this device?

IgnacioPuchet, 2025-07-12:
Of course, this is just before it leaves the network
`{"id":196,"type":"EndDevice","ieeeAddr":"0xffffb40e06059384","nwkAddr":62557,"manufId":5127,"manufName":"Third Reality, Inc","powerSource":"Battery","modelId":"3RSM0147Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,65281,1,1026,1029],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":16,"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"swBuildId":"v1.00.38","hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":38,"stackVersion":2,"hwVersion":0,"swBuildId":"v1.00.38","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1752296767859}`

Koenkk, 2025-07-12:
@3reality-support is seems the sensor is missing the soil moisture cluster (`0x0408`), is that expected?

3reality-support, 2025-07-14:
Yes, 0x0408 has not been registered in the current version yet and will be fixed in the next version

IgnacioPuchet, 2025-07-14:
Is there a way to avoid the device from leaving the network so I can use the humidity sensor until it's fixed?

IgnacioPuchet, 2025-07-18:
The device leaving the network didn't seem to be related to the sensor configuration failure, I was able to fix that by updating the coordinator firmware to version 20240710. 

 The device now stays in the network with no value in the soil moisture sensor 
[image]