Koenkk/zigbee2mqtt #27992: Throttle setting does not work (as well as debounce) - messages still published

hnykda, 2025-07-14:
### What happened?

Hey. I have a dimmer: TS0601_dimmer_1_gang_1. It produces too many messages, such as this on a single short touch:

<details>

```

zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:36+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:36+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:37+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:37+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:37+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:37+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:37+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:38+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:38+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:38+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:38+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:38+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:39+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:39+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:39+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:39+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:39+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:40+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:40+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:40+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:40+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:40+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:41+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:41+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:41+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}

zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:44+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}

zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:44+02:00","light_type":"led","linkquality":69,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
zigbee2mqtt/DimmerKitchen {"backlight_mode":"normal","brightness":254,"countdown":0,"last_seen":"2025-07-14T23:26:44+02:00","light_type":"led","linkquality":65,"max_brightness":254,"min_brightness":23,"power_on_behavior":"previous","state":"ON"}
zigbee2mqtt/DimmerKitchen/type {"light":"led"}
[... cut ...] utes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_w4cryh2i","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1752526315367}

Nerivec, 2025-07-15:
Throttle/debounce require a restart of Z2M to be effective. I assume you did?

hnykda, 2025-07-15:
Oh, embarrassing. I think that should be mentioned in the docs. Sorry...