Koenkk/zigbee2mqtt #27996: [New device support]: ZMS-206US-1

villifesto, 2025-07-15:
### Link

https://www.zemismart.com/products/zms-206us?srsltid=AfmBOoq-6cXhedS2o1PaeT2hyQTiUEKQUIvCrnqvGKePf5LwWSmCjftP&VariantsId=17620

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0xa4c1383c09042a03","nwkAddr":29661,"manufId":4417,"manufName":"_TZE204_sa2ueffe","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�3\t0f�3\t0\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1752559184196}

### Zigbee2MQTT version

2.4.0

### Comments

Hi,
This is the 1 gang version of TS0601, and it shows up as "Not supported".  

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_sa2ueffe',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

[image]

Koenkk, 2025-07-15:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE284_3ctwoaip ` to `_TZE284_nhgdf6qr`
- Start Z2M and see if the device now works

villifesto, 2025-07-18:
Done. Thank you.