Koenkk/zigbee2mqtt #28002: Regular crashes and impossible to remove devices

mbrucher, 2025-07-15:
### What happened?

Since adding new thermostats that were configured 6 months earlier, the network is completely unstable and zigbee2mqtt crashes every couple of hours. After the OTA Philips debacle, I have tried to remove them from the network, but they still appear in Mosquitto MQTT.

Errors failing to remove devices are looking like these:
[Failed to remove rejected device](zh:controller: Failed to remove rejected device: Delivery failed for '16352'.)

There are recurrent issues with Third Reality thermostats
z2m: Failed to configure '0x282c02bfffeaae89', attempt 3 (Error: Bind 0x282c02bfffeaae89/1 msTemperatureMeasurement from '0xbc026efffe24c587/1' failed (Delivery failed for '62517'.)

Crashes are of the form:
[2025-07-15 20:03:56] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-07-15 20:03:56] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR

I'm seriously considering reseting everything, but I'm not even sure this will be enough or how to do this, as Mosquitto seems to have old stale data, but mostly reseting some of the light systems means accessing the electrical board.

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

7.4.3

### Adapter

ZBT-1

### Setup

HomeAssistant Green

Nerivec, 2025-07-16:
Before going too far down the rabbit hole, try to update your coordinator's firmware to 8.0.2, it include stability improvements around this kind of issue.
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update7

Also make sure you don't have too much interference (USB or ZigBee).
https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html#usb-based-adapter

mbrucher, 2025-07-16:
Thanks, I'm going to try this then over the weekend when I can carefully analyse also your suggestions, thanks!

closed 2025-09-22 by github-actions[bot]