Koenkk/zigbee2mqtt #28005: Zigbee2MQTT exists with code 0 on fatal error

NyCodeGHG, 2025-07-15:
### What happened?

Zigbee2MQTT doesn't exit with a non-zero exit code on a fatal adapter error.

### What did you expect to happen?

Zigbee2MQTT should exit with a non-zero exit code on fatal errors.

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

NixOS

Nerivec, 2025-07-16:
That would be the `Adapter disconnected, stopping` path, correct?

@Koenkk maybe we should assign an override exit code for this code path? For good measure, a unique one that's specific to "adapter disconnected" (all current, that use `1`, are request failures).
https://github.com/Koenkk/zigbee2mqtt/blob/ffdd353e16e6e6170942070c2112cffbe6afea9b/lib/controller.ts#L321-L324
_I guess quite a few versions ago it always exited with `1` because `this.zigbee.stop` would always fail since it tried to stop join/backup (which no longer happens since now it executes cleanly by avoiding "doomed to fail" requests)._

Koenkk, 2025-07-18:
Fixed in https://github.com/Koenkk/zigbee2mqtt/pull/28013

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-07-18 by Koenkk