Koenkk/zigbee2mqtt #28010: TS011F_plug_1: child_lock gets removed when pressing the on/off-button quickly

hirtema, 2025-07-16:
### What happened?

This is something i found out while investigating issue #27963 : when testing new bought sockets i found out that the "child_lock" is not holding steady on TS011F_plug_3, and later also on the TS011F_plug_1 type of socket. For TS011F_plug_3 see mentioned issue (i found a workaround i think), this issue is about the TS011F_plug_1.

When pressing the on/off-button on the socket slowly, like every 2-3 seconds, the "child_lock" state stays set and works as intended. But when pressing the button several times per second, a set "child_lock" gets removed after around 3-5 presses. The "child_lock" slider in the configuration page doesn't reflect the new state of the lock (don't know if it should also be updated like the power switch slider).

When having the "child_lock" already set in the device configuration at Z2M startup, the behaviour is the same (no workaround like on the TS011F_plug_3 with restarting Z2M to make the child_lock persistant, will have another look at the problem on that device later)

### What did you expect to happen?

A "child_lock" set in the device configuration should be persistant and not removeable by pressing the on/off-button on the device quickly.

### How to reproduce it (minimal and precise)

Setup and configure a "TS011F_plug_1"-socket (on/off indicator mode, power-on state, ...)
Set the "child_lock" in the device settings to "locked", confirm the change on the bottom of the settings page

Press the on/off-button on the socket **slowly** (once every 2-3 seconds), the device does not switch as intended
Press the on/off-button on the socket **several times per second**, after around 3-5 presses the "child_lock" state breaks, the devices switches again when actuated locally

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

EmberZNet 8.0.2 [GA]

### Adapter

SLZB-06M with ZigBee-Coordinator-Firmware 20250220 (EFR32MG21 radio)

### Setup

Z2M on HomeAssistant 16.1 OS in QEMU (QNAP Virtualization Station)

Issue originally noticed on a network running HA 15.2 in a ProxMox VM with a SONOFF USB Dongle Plus coordinator, same Z2M-version

### Debug log

The log starts with setting the lock state to "locked" in the Z2M device settings page. Dots are cleaned out messages of other devices. If further logging is required please contact me.

[TS011F_plug_1_child_unlock.txt](https://github.com/user-attachments/files/21262386/TS011F_plug_1_child_unlock.txt)

lux73, 2025-07-18:
Sounds to me for a Tuya Firmware Bug which is not fixable by z2m

hirtema, 2025-07-19:
@lux73 since it's Tuya very likely can be. I also thought about it and how to check it.

So today i set the lock, checked it for being set and then killed power to the coordinator, Z2M and the network then were out of the picture, and the device was on it's own. And guess what: when pressing the button quickly, the child lock unlocked again, slow presses didn't break it.

So i think you might be right that on this device it's a firmware bug (my sockets all have version 192, which is the latest). Since i discovered a similar problem on the TS011F_plug_3, which went away after restarting Z2M, i hoped it would be the same here and Z2M maybe is involved.

If the TS011F_plug_1 indeed has a built-in bug here, maybe Z2M can warn on the device page about the child_lock not working properly at the moment. But maybe someone else can also reproduce it, this type of device should be very common, it's the round version of the Tuya plug, while the cube-like Tuya plug usually seems to be a TS011F_plug_3.

benn447, 2025-08-05:
I was reading through the zigbee2mqtt documentation today and stumbled across this "To reset Child Lock without Z2M, quickly press the physical button 4 times" apparently it's a documented feature?

https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#reset-lock

hirtema, 2025-08-28:
Ah, thanks for finding that. So it works as designed (but not the way i expected a child lock to work).

closed 2025-08-28 by hirtema