Koenkk/zigbee2mqtt #28012: [New device support]:

nlochschmidt, 2025-07-16:
### Link

https://kb.shelly.cloud/knowledge-base/shelly-2pm-gen4

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x7c2c67fffe7af7d4","nwkAddr":62084,"manufId":5264,"manufName":"Shelly","powerSource":"Unknown","modelId":"2PM","epList":[1,239,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x980c33fffe820140","endpointID":1}],"configuredReportings":[],"meta":{}},"239":{"profId":49153,"epId":239,"devId":8193,"inClusterList":[64513,64514],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1752699583552}

### Zigbee2MQTT version

2.5.1

### Comments

Received a Shelly 2PM Gen 4 today. Device model in the Shelly docs is S4SW-002P16EU. I am running the Shelly in the device profile "cover".

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['2PM'],
    model: 'S4SW-002P16EU',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"239":239}}), m.windowCovering({"controls":["lift","tilt"]})],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

I'll try with actual motorized shutters soon, but right now I could only do a test run without a connected motor. 

What works: 
- Switching between Open, Close and Stop activates the relays
- Selecting a position or tilt activates the relays as well

What doesn't work:
- The energy measurements aren't exposed. In the Shelly app I can see power, voltage, current, frequency and energy readings
- I can't tell if the specific settings "Invert Cover" and "Cover position tilt disable report" do anything. I probably have to calibrate the Shelly first?

I am happy to provide any additional information or run tests as needed.

lukasz-burak, 2025-07-17:
To add to that, it should also support switch mode, not just windowCover

`import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['2PM'],
    model: '2PM',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"239":239}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};
`

nlochschmidt, 2025-07-17:
@lukasz-burak absolutely! I changed my 2PM to "switch" mode, but was unable to get measurements. Did it work for you? 

I wonder how to make the definition dynamic based on the configured mode ("switch" or "cover")?

hhromic, 2025-07-22:
According to the [official documentation](https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/DeviceFeatures), the 2PM Gen4 in both, Cover and Switch modes, does not seem to expose energy measurement over standard Zigbee metering/measurement clusters (`0x0702` or `0x0B04`). Contrary to the 1PM (mini) devices.

Perhaps the Shelly app is able to obtain those via the Shelly RPC channel cluster?

Hope Shelly updates the firmware to support standard clusters though, like in the 1PM Gen4.

jfmlima, 2025-08-12:
Hey @Koenkk, this device currently supports 2 modes, switch and cover.
Is a setup like this supported? Basically picking up the corrected config based on the selected mode.

Here's the generated config when in each mode (selecting modes in shelly UI).

```ts
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['2PM'],
    model: '2PM',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"239":239}}), m.windowCovering({"controls":["lift","tilt"]})],
    meta: {"multiEndpoint":true},
};
```

```ts
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['2PM'],
    model: '2PM',
    vendor: 'Shelly',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"239":239}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};
```

This is still pending, but probably related https://github.com/Koenkk/zigbee-herdsman-converters/pull/9468.

jfmlima, 2025-08-14:
> [@jfmlima](https://github.com/jfmlima)
> 
> * when changing the switch mode, do you re-pair the device afterwards?
> * Could you provide the data/database.db entry of both cases?

@Koenkk 
- I'll have to double-check that but I know for sure they are handled as 2 different devices with different identifiers even though it's the same physical device.

Here's the database.db (first one is cover mode, second is switch)
```json
[
    {
        "id": 27,
        "type": "Router",
        "ieeeAddr": "0x7c2c67fffexxxxxx",
        "nwkAddr": 21384,
        "manufId": 5264,
        "manufName": "Shelly",
        "powerSource": "Unknown",
        "modelId": "2PM",
        "epList": [
            1,
            239,
            242
        ],
        "endpoints": {
            "1": {
                "profId": 260,
                "epId": 1,
                "devId": 514,
                "inClusterList": [
                    0,
                    3,
                    4,
                    5,
                    258
                ],
                "outClusterList": [],
                "clusters": {
                    "genBasic": {
                        "attributes": {
                            "modelId": "2PM",
                            "manufacturerName": "Shelly",
                            "powerSource": 0,
                            "zclVersion": 8
                        }
                    },
                    "closuresWindowCovering": {
                        "attributes": {
                            "currentPositionLiftPercentage": 100,
                            "currentPositionTiltPercentage": 50,
                            "configStatus": 3
                        }
                    }
                },
                "binds": [
                    {
                        "cluster": 258,
                        "type": "endpoint",
                        "deviceIeeeAddress": "0x84b4dbfffebcxxx",
                        "endpointID": 1
                    }
                ],
                "configuredReportings": [
                    {
                        "cluster": 258,
                        "attrId": 8,
                        "minRepIntval": 1,
                        "maxRepIntval": 65000,
                        "repChange": 1
[... 268 lines cut ...]
                "configuredReportings": [],
                "meta": {}
            },
            "242": {
                "profId": 41440,
                "epId": 242,
                "devId": 97,
                "inClusterList": [],
                "outClusterList": [
                    33
                ],
                "clusters": {},
                "binds": [],
                "configuredReportings": [],
                "meta": {}
            }
        },
        "zclVersion": 8,
        "interviewCompleted": true,
        "interviewState": "SUCCESSFUL",
        "meta": {},
        "lastSeen": 1755127591272
    }
]
```

nlochschmidt, 2025-08-14:
@Koenkk 
> when changing the switch mode, do you re-pair the device afterwards?

When switching the profile the device leaves the Zigbee network. It gets automatically removed from zigbee2mqtt. The device then reboots and goes into pairing mode. After joining again, the IEEE address is incremented by 1. 

When switching back to cover mode it does the reverse and restores the original address again.

Unfortunately the modelId stays the same in both modes `2PM`.

Koenkk, 2025-08-14:
@nlochschmidt can you compare the db entry before and after? Maybe something changes in e.g. the `inClusterList`

nlochschmidt, 2025-08-14:
@Koenkk yes, see the example @jfmlima [shared above](https://github.com/Koenkk/zigbee2mqtt/issues/28012#issuecomment-3186264381). Not sure how that can be used in the converter though.

jfmlima, 2025-08-14:
@nlochschmidt @Koenkk worth mentioning that the devices I added here from the database.db are different as I already had them connected to my instance.
I can share the same device entries in the db if needed later today.

Kranzes, 2025-08-20:
> I also tried to perform the calibration via the web server first, which works, but the calibration data is discarded when switching to Zigbee mode...

I haven't got my unit yet, but could you contact Shelly regarding this?

jfmlima, 2025-08-22:
> Thanks [@jfmlima](https://github.com/jfmlima) !
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Sounds good, thanks @Koenkk! 
On the documentation side, do I need to make any change? Adding images based on the new names perhaps? 
When it's out I'll also add some notes there with information about changing between modes.

ahemwe, 2025-08-26:
Slightly off topic: I've added m Shelly 2PM Gen4 to z2m, but I lack a remote switch that can be **bound** to the Shelly. This would be the most relevant argument for using zigbee instead of matter since it's controller-free.
Any recommedations? I tried a simple https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743 which can be bound, but shows no functionality.

Kranzes, 2025-08-26:
> > I also tried to perform the calibration via the web server first, which works, but the calibration data is discarded when switching to Zigbee mode...
> 
> [@Ulf92](https://github.com/Ulf92) I haven't got my unit yet, but could you contact Shelly regarding this?

Update: calibration works fine for me and i can initialize it in zigbee mode through the web UI of the Shelly.

ThatMishakov, 2025-08-29:
I have this relay in switch mode, on 1.7.0, and the energy metering is not working for me (which isn't a big deal for me). I am also getting constant errors like this with it (about once per minute)
```
info 2025-08-29 01:41:27zh:ember:ezsp: Received network/route error ROUTE_ERROR_NON_TREE_LINK_FAILURE for "61544".
info 2025-08-29 01:41:52zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "61544".
```
Is there anything I can do to debug it?

majd, 2025-09-01:
@Koenkk After the most recent update, Z2M correctly recognizes the module in switch mode. However, the state only seem to be reflected in Z2M frontend (and other MQTT integrations) when the switch is switched through Z2M. If I physically toggle the switch, the state does not get reflected in Z2M.

I checked the logs and I do see the state updates there when I physically use the switch.

> [9/1/2025, 11:21:44 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Wall Switch', payload '{"ac_frequency":49.93,"ac_frequency_l1":null,"ac_frequency_l2":null,"current":0.011,"current_l1":null,"current_l2":null,"energy":0,"energy_l1":null,"energy_l2":null,"last_seen":"2025-09-01T21:21:44.881Z","linkquality":148,"power":0,"power_apparent":0,"power_factor":0,"power_l1":null,"power_l2":null,"produced_energy":0,"produced_energy_l1":null,"produced_energy_l2":null,"state":"OFF","state_l1":"OFF","state_l2":"OFF","voltage":228.01,"voltage_l1":null,"voltage_l2":null}'
>
> [9/1/2025, 11:21:45 PM] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Wall Switch', payload '{"ac_frequency":49.93,"ac_frequency_l1":null,"ac_frequency_l2":null,"current":0.011,"current_l1":null,"current_l2":null,"energy":0,"energy_l1":null,"energy_l2":null,"last_seen":"2025-09-01T21:21:45.169Z","linkquality":184,"power":0,"power_apparent":0,"power_factor":0,"power_l1":null,"power_l2":null,"produced_energy":0,"produced_energy_l1":null,"produced_energy_l2":null,"state":"ON","state_l1":"OFF","state_l2":"OFF","voltage":228.01,"voltage_l1":null,"voltage_l2":null}'

jfmlima, 2025-09-01:
> [@Koenkk](https://github.com/Koenkk) After the most recent update, Z2M correctly recognizes the module in switch mode. However, the state only seem to be reflected in Z2M frontend (and other MQTT [...]

I was able to replicate, it was pointing to the incorrect endpoints. 
Here's the fix https://github.com/Koenkk/zigbee-herdsman-converters/pull/9905, @Koenkk can we patch this release?
It's non-critical, but would be nice to have proper readings 👍 

They also updated the documentation which is nice https://shelly-api-docs.shelly.cloud/gen2/Integrations/Zigbee/DeviceFeatures

dekiesel, 2025-09-15:
I updated to the `latest-dev` since the PR was merged there in hopes of being able to set certain percentages ( I use my 2pm as a cover). Unfortunately the slider doesn't work and when I, f. ex., click on the 25 percent button the cover closes all the way

[image]

Is that expected behaviour at the moment or am I doing something wrong?

dekiesel, 2025-09-15:
Thanks, yes, I did. I just switched back to wifi and recalibrated, but if you think it makes a difference I'll try again?

jfmlima, 2025-09-15:
> Thanks, yes, I did. I just switched back to wifi and recalibrated, but if you think it makes a difference I'll try again?

I think you need to calibrate in zigbee mode as switching to wifi will reset calibration. 
Can you try switching to zigbee, calibrate and try again, please?

jfmlima, 2025-09-15:
> [@jfmlima](https://github.com/jfmlima) That worked, thank you!
> 
> Would it be a good idea to turn off wifi now? Or should I wait until things are a bit more stable?

I still have wifi enabled, just disconnected from their cloud services, as not all actions are supported through Zigbee, yet.

NetRat88, 2025-09-17:
Hi together,

any ideas why one of my 14 x 2PM Gen4 devices in Cover mode don't have the vendor and correct model assigened?

[image]

I already tried to delete the device in my Home Assistant Zigbee2MQTT and als made a factory reset but it is always added without the correct vendor and model. Another difference between the working shellys is the status:

the not correctly assigned one:
`{
    "linkquality": 144,
    "position": 84,
    "position_1": 84,
    "state_1": "OPEN",
    "tilt_1": 50
}`

the correct assigned ones:
`{
    "linkquality": 128,
    "position": 50,
    "state": "OPEN",
    "tilt": 50
}`

I am still using the stable version:

[image]

Many thanks in advance and best regards

ggmanugg, 2025-09-24:
Mine also started working correctly recently :)