Koenkk/zigbee2mqtt #28017: [New device support]: Tuya ZTH05

7-ich, 2025-07-17:
### Link

https://de.aliexpress.com/item/1005008919417924.html?spm=a2g0o.order_list.order_list_main.5.17d85c5f55XUyi&gatewayAdapt=glo2deu

### Database entry

'{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601","options":[],"supports_ota":false,"vendor":"_TZE284_upagmta9"},"friendly_name":"Tuya","ieee_address":"0xa4c138fce81c8421","status":"successful","supported":false},"type":"device_interview"}'

### Zigbee2MQTT version

2.5.1-1

### Comments

Unfortunately, the Tuya ZTH05 sensor is not recognized by zigbee2mqtt. The status says "not supported," even with the external definition.

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['3RSPE01044BZ'],
    model: '3RSPE01044BZ',
    vendor: 'Third Reality, Inc',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I created a JS file in external_converter. I have gotten Tuya recognized to the point that it is displayed under Devices with a logo, but without exposes (No temperature, no humidity, no battery state). 

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_upagmta9',
        },
    ],
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Tuya temperature/humidity sensor (ZTH05)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(),
        e.humidity(),
        tuya.exposes.batteryState()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [3, 'battery_state', tuya.valueConverter.batteryState],
        ],
    },
};

module.exports = definition;
```

[image]

[image]

Koenkk, 2025-07-18:
`_TZE284_upagmta9` is supported in the dev branch already

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-07-18 by Koenkk

7-ich, 2025-07-18:
Thanks very much for your work! 👌

7-ich, 2025-08-03:
Hi @Koenkk, 

The Zigbee sensor is now recognized directly. However, it still doesn't display any data.

Any Idea?

[image]