Koenkk/zigbee2mqtt #28018: Can´t find any Device

AvooCado, 2025-07-17:
### What happened?

Rebuild my server and installed a fresh Docker container. Updated my Sonoff USB Dongle P with the latest firmware. But can´t seem to find any device.

### What did you expect to happen?

Find Devices.

### How to reproduce it (minimal and precise)

Just start container and wait to find devices.

### Zigbee2MQTT version

2.5.1-dev commit: 21fa5cbdeddb51d55364c4d28b887791384a951b

### Adapter firmware version

20250321

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus -P

### Setup

On Unraid in a docker container

### Debug log

Debug at the start of searching for new devices:
`[2025-07-17 22:05:17] info:     z2m: Zigbee: allowing new devices to join.
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:writer: --> frame [254,5,37,54,15,252,255,254,1,229]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: <-- [254,1,101,54,0,82,254,3,69,182,0,0,0,240]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,101,54,0,82,254,3,69,182,0,0,0,240]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 54 - [0] - 82
[2025-07-17 22:05:17] debug:    zh:zstack:znp: <-- SRSP: ZDO - mgmtPermitJoinReq
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,69,182,0,0,0,240]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
[2025-07-17 22:05:17] debug:    zh:zstack:znp: <-- AREQ: ZDO - mgmtPermitJoinRsp
[2025-07-17 22:05:17] debug:    zh:controller: Received ZDO response: clusterId=PERMIT_JOINING_RESPONSE, status=SUCCESS, payload=undefined
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-07-17 22:05:17] debug:    zh:zstack:znp: --> SREQ: AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":1,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,2,2,11,254,0]}}
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:writer: --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,1,0,30,6,0,25,2,2,11,254,0,232]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2025-07-17 22:05:17] debug:    zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: <-- [254,3,68,128,225,242,1,213]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext [254,3,68,128,225,242,1,213]
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [225,242,1] - 213
[2025-07-17 22:05:17] debug:    zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":225,"endpoint":242,"transid":1}
[2025-07-17 22:05:17] debug:    zh:zstack:unpi:parser: --- parseNext []
[2025-07-17 22:05:17] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"u0d4h-1"}'
[2025-07-17 22:09:32] debug:    z2m: Saving state to file /app/data/state.json
[2025-07-17 22:14:32] debug:    z2m: Saving state to file /app/data/state.json`

mikki8, 2025-07-18:
Have the same issue. Same setup but not dev. Can't add any devices anymore

mikki8, 2025-07-24:
I can add usb switches. It seems like I have a Problem to add just a noua a1z.

mikebzh44, 2025-07-25:
Same for me with a fresh installation

2.5.1 : I can't add new devices (tested with a AQARA double rocker)

So I tested with a 2.4.0 installed on a RPi 4 and the same device is paired at the first try

Hope 2.5.2 will fix this bug

Koenkk, 2025-07-25:
- Can you try enabling joining on just the coordinator and try pairing via that?
- If that doesn't help, can you do a [coordinator check](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check)?

mikebzh44, 2025-07-25:
Sorry but I have downgrade to 2.4.0 to restore the nehavior of my Home Assistant and don't bother my wife :D

mikki8, 2025-07-26:
hey how can you downgrade it? looked in the addon in home assistant couldnt find it

mikebzh44, 2025-07-26:
It's because I'm running Z2M and Mosquitto on a dedicate machine (Orange Pi Zero) so I can install any version ;)

closed 2025-10-02 by github-actions[bot]