Koenkk/zigbee2mqtt #28024: Unable to control water valve status for TS0601_water_switch

adam900710, 2025-07-19:
### What happened?

The support for the device is added in v2.4.0, and it is properly published to home-assistant or the web front end.

But I failed to control the valves, I can switch it on off, but it just has no effect.

Tried both 2.4.0 and the latest 2.5.1, no difference, can pair and detect the device, just read-only values, no real on/off control on the valves.

### What did you expect to happen?

I can control the valve status (on/off) correctly.

### How to reproduce it (minimal and precise)

1. Connect the device to z2m
2. Enable webUI
3. Use WebUI to control the valve (on/off)

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

Unrelated

### Adapter

Unrelated

### Setup

Plain z2m installation on Archlinux, on a x86_64 machine.

### Device `database.db` entry

{"id":6,"type":"EndDevice","ieeeAddr":"0x30fb10fffe164d4e","nwkAddr":58230,"manufId":4098,"manufName":"_TZE284_eaet5qt5","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE284_eaet5qt5","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1752902805159}

### Debug log

```
Jul 19 14:56:40 nuc node[36179]: [2025-07-19 14:56:40] info:         zh:controller: Interview for '0x30fb10fffe164d4e' started
Jul 19 14:56:40 nuc node[36179]: [2025-07-19 14:56:40] info:         z2m: Starting interview of '0x30fb10fffe164d4e'
Jul 19 14:56:40 nuc node[36179]: [2025-07-19 14:56:40] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x30fb10fffe164d4e","ieee_address":"0x30fb10fffe164d4e","status":"started"},"type":"device_interview"}'
Jul 19 14:56:40 nuc node[36179]: [2025-07-19 14:56:40] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x30fb10fffe164d4e","ieee_address":"0x30fb10fffe164d4e"},"type":"device_announce"}'
Jul 19 14:56:44 nuc node[36179]: [2025-07-19 14:56:44] info:         zh:controller: Succesfully interviewed '0x30fb10fffe164d4e'
Jul 19 14:56:44 nuc node[36179]: [2025-07-19 14:56:44] info:         z2m: Successfully interviewed '0x30fb10fffe164d4e', device has successfully been paired
Jul 19 14:56:44 nuc node[36179]: [2025-07-19 14:56:44] info:         z2m: Device '0x30fb10fffe164d4e' is supported, identified as: Tuya Dual water valve (TS0601_water_switch)
Jul 19 14:56:44 nuc node[36179]: [2025-07-19 14:56:44] info:         z2m: Configuring '0x30fb10fffe164d4e'
Jul 19 14:56:44 nuc node[36179]: [2025-07-19 14:56:44] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Dual water valve","exposes":[{"access":1,"description":"Valve 1 status (manual, auto, idle)","endpoint":"l1","label":"Valve status","name":"valve_status","property":"valve_status_l1","type":"enum","values":["manual","auto","idle"]},{"access":1,"description":"Valve 2 status (manual, auto, idle)","endpoint":"l2","label":"Valve status","name":"valve_status","property":"valve_status_l2","type":"enum","values":["manual","auto","idle"]},{"description":"Valve 1 on/off","endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","label":"State","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"label":"Valve 1","type":"switch"},{"description":"Valve 2 on/off","endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","label":"State","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"label":"Valve 2","type":"switch"},{"access":3,"description":"Valve 1 countdown in minutes","endpoint":"l1","label":"Countdown","name":"countdown","property":"countdown_l1","type":"numeric","unit":"min","value_max":1440,"value_min":0},{"access":3,"description":"Valve 2 countdown in minutes","endpoint":"l2","label":"Countdown","name":"countdown","property":"countdown_l2","type":"numeric","unit":"min","value_max":1440,"value_min":0},{"access":1,"description":"Valve 1 irrigation last duration in seconds","endpoint":"l1","label":"Valve duration","name":"valve_duration","property":"valve_duration_l1","type":"numeric","unit":"s"},{"access":1,"description":"Valve 2 irrigation last duration in seconds","endpoint":"l2","label":"Valve duration","name":"valve_duration","property":"valve_duration_l2","type":"numeric","unit":"s"},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_water_switch","options":[],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0x30fb10fffe164d4e","ieee_address":"0x30fb10fffe164d4e","status":"successful","supported":true},"type":"device_interview"}'
Jul 19 14:56:45 nuc node[36179]: [2025-07-19 14:56:45] info:         z2m: Successfully configured '0x30fb10fffe164d4e'
Jul 19 14:56:52 nuc node[36179]: [2025-07-19 14:56:52] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe164d4e', payload '{"linkquality":148,"state_l1":"ON"}'
Jul 19 14:56:54 nuc node[36179]: [2025-07-19 14:56:54] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe164d4e', payload '{"linkquality":148,"state_l1":"ON","state_l2":"ON"}'
Jul 19 14:57:12 nuc node[36179]: [2025-07-19 14:57:12] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1389b3a8f3169', payload '{"battery":100,"humidity":62,"humidity_calibration":0,"linkquality":184,"temperature":14.9,"temperature_calibration":0,"temperature_unit":"celsius"}'
Jul 19 14:57:12 nuc node[36179]: [2025-07-19 14:57:12] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1389b3a8f3169', payload '{"battery":100,"humidity":65,"humidity_calibration":0,"linkquality":180,"temperature":14.9,"temperature_calibration":0,"temperature_unit":"celsius"}'
Jul 19 14:57:14 nuc node[36179]: [2025-07-19 14:57:14] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe164d4e', payload '{"linkquality":148,"state_l1":"OFF","state_l2":"ON"}'
Jul 19 14:57:15 nuc node[36179]: [2025-07-19 14:57:15] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe164d4e', payload '{"linkquality":148,"state_l1":"OFF","state_l2":"OFF"}'
Jul 19 14:57:18 nuc node[36179]: [2025-07-19 14:57:18] info:         z2m: Zigbee: disabling joining new devices.
Jul 19 14:57:19 nuc node[36179]: [2025-07-19 14:57:19] info:         zh:ember: [STACK STATUS] Network closed.
Jul 19 14:57:19 nuc node[36179]: [2025-07-19 14:57:19] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":0},"status":"ok","transaction":"mj201-2"}'
```

Koenkk, 2025-07-19:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/27833 and fixed already in the dev branch.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-07-19 by Koenkk

adam900710, 2025-07-19:
Verified the fix is working, I just applied the diff into`/usr/share/webapps/zigbee2mqtt/node_modules/zigbee-herdsman-converters/dist/devices/tuya.js` and now it works fine.