Koenkk/zigbee2mqtt #28035: [New device support]: _TZE204_gops3slb Floor Heating Thermostat

HD78, 2025-07-20:
### Link

https://www.aliexpress.com/item/1005006457769958.html

### Database entry

{"id":66,"type":"Router","ieeeAddr":"0xa4c138c18cbd501e","nwkAddr":49524,"manufId":4417,"manufName":"_TZE204_gops3slb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u000f0f��\u000f0\u0012��\u000f0e��\u000f0e��\u000f0e","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_gops3slb","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1753005748865}

### Zigbee2MQTT version

2.5.1

### Comments

I have had couple of these devices with id _TZE204_oh8y8pv8 and they have worked fine.
But now new ones came with id _TZE204_gops3slb. 

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_gops3slb',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Adding that and making this external converter it works:

```
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const color = require("zigbee-herdsman-converters/lib/color");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const e = exposes.presets;
const ea = exposes.access;
const legacy = require("zigbee-herdsman-converters/lib/legacy");
const lumi = require("zigbee-herdsman-converters/lib/lumi");
const utils = require("zigbee-herdsman-converters/lib/utils");
const {logger} = require("zigbee-herdsman-converters/lib/logger");
const {develcoModernExtend} = require("zigbee-herdsman-converters/lib/develco");
const m = require("zigbee-herdsman-converters/lib/modernExtend");

const definition = {
    zigbeeModel: ["TS0601"],
    fingerprint: tuya.fingerprint("TS0601", ["_TZE200_viy9ihs7", "_TZE204_lzriup1j", "_TZE204_xnbkhhdr", "_TZE284_xnbkhhdr", "_TZE204_oh8y8pv8", "_TZE204_gops3slb"]),
    model: "ZWT198/ZWT100-BH",
    vendor: "Tuya",
    description: "Wall thermostat",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary("factory_reset", ea.STATE_SET, "ON", "OFF").withDescription("Full factory reset, use with caution!"),
        e.child_lock(),
        e.temperature_sensor_select(["internal", "external", "both"]),
        e
            .climate()
            .withSystemMode(["off", "heat"], ea.STATE_SET)
            .withPreset(["manual", "auto", "temporary_manual"])
            .withSetpoint("current_heating_setpoint", 5, 35, 0.5, ea.STATE_SET)
            .withRunningState(["idle", "heat"], ea.STATE)
            .withLocalTemperature(ea.STATE)
[... 112 lines cut ...]
            [110, "backlight_mode", tuya.valueConverter.backlightModeOffLowMediumHigh],
            // ============== found but not functional datapoints:

            // [16, 'min_temperature_limit', tuya.valueConverter.divideBy10],  // datapoint listed in Tuya, but no communication from device
            // [105, 'dp105', tuya.valueConverter.onOff],                      // not listed in Tuya, but device sends datapoint
            // [111, 'dp111', tuya.valueConverter.onOff],                      // not listed in Tuya, but device sends datapoint

            // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
            // For each period:
            // 1st byte: hour
            // 2nd byte: minute
            // 3rd, 4th bytes: temperature multiplied by 10
            // On the device last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
            // scheduling can't be configured at all on the device.
            // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
            // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
            // Then the schedule is:
            // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
            // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
        ],
    },
};

module.exports = definition;
```

HD78, 2025-07-22:
Also found out that the auto/manual modes are reversed with this. So changed this:
```
                    if (device.manufacturerName === "_TZE204_lzriup1j") {
                        return {
                            auto: tuya.enum(1),
                            manual: tuya.enum(0),
                            temporary_manual: tuya.enum(2),
                        };
```
To this:
```
                    if (device.manufacturerName === "_TZE204_gops3slb") {
                        return {
                            auto: tuya.enum(1),
                            manual: tuya.enum(0),
                            temporary_manual: tuya.enum(2),
                        };
```
So they are the right way around for this model.

Koenkk, 2025-08-04:
Thanks @bernardesarthur !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

0sy, 2025-09-28:
Hello - thank you for adding support for this device. I bought one to make sure it works ok, and it did.

Now, I'm not sure what triggers this, but I've started having an issue today.

Controlling it HA (via Z2M addon) works as expected, no issues or errors / warnings in logs.

So that's fine.

However, when physically trying to operate it from the touch screen, whenever you use the "up arrow" to try and adjust temperature upwards, the temp value goes to 0.00 C.

If you try to tap the up arrow again, nothing happens, it remains at 0.00. Using the "down arrow" starts the temp from 99.99 C, and you can repeatedly hammer it to go down to your desired value (even though it takes a lot of presses, as the increment is 0.5).

However, as soon as you press "UP" again, it goes to 0.00 C again, and you need to start over.

(During all this, in HA the target temp is shows as ~6000 Celsius, although you can adjust it from HA back to normal in one action)

I have tried cutting power to the device for 1-2 mins - did not help.

Adjusting from HA works as intended, manually operating the panel for adjust temp does not.

0sy, 2025-09-29:
@HD78 - can you check the above please?