Koenkk/zigbee2mqtt #28038: [New device support]: _TZ3000_axpdxqgu

ustari28, 2025-07-20:
### Link

https://moeshouse.com/products/eu-star-ring-smart-light-switch

### Database entry

EndDevice 0xEA8A Network address in decimal format: 60042 Last seen N/A Availability: Disabled Power 84% Battery Firmware ID Unknown Unknown ZigBee Model TS0041 _TZ3000_axpdxqgu (Wireless switch with 1 button) Model TS0041 Tuya

### Zigbee2MQTT version

2.2.0

### Comments

In previous versions worked.

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0041'],
    model: 'TS0041',
    vendor: '_TZ3000_axpdxqgu',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.onOff({"powerOnBehavior":false})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

No action available when you press the button

Koenkk, 2025-08-20:
> No action available when you press the button

But I see that an action is published, am I missing something?

```
[2025-08-19 14:44:01] info: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ligh_2_dressing_room', payload '{"action":"single","battery":100,"linkquality":134,"voltage":3000}'
```

Koenkk, 2025-08-20:
That's because you enabled the `legacy_action_sensor` ([docs](https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html))

ustari28, 2025-08-28:
Ok, I changed to false  then I can see the "single" message in action but I still have no toggle or button on/off.

Koenkk, 2025-08-28:
>  but I still have no toggle or button on/off.

This device does not have a light directly connected to it right? In that case it is expected that it has no on/off switch, it only send an event when pressed.