Koenkk/zigbee2mqtt #28040: Lidl lightstrip gets wrong property in Home assistant. Typo.

FransHalsjes, 2025-07-21:
### What happened?

When integrating the Lidl Lightstip:
TS0505A
_TZ3000_riwp3k79 (Livarno Home RGB+CCT LED light strip 2m)

With Home asssistant, I get the following error in HA:

Logger: homeassistant.components.mqtt.select
Bron: components/mqtt/select.py:125
integratie: MQTT ([documentatie](https://www.home-assistant.io/integrations/mqtt), [problemen](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
Eerst voorgekomen: 20 juli 2025 om 02:10:37 (3 gebeurtenissen)
Laatst gelogd: 20 juli 2025 om 20:53:09

Invalid option for select.bar_lichtstrip_color_power_on_behavior: 'cutomized' (valid options: ['initial', 'previous', 'customized'])

This is obviously a typo, as it says "cutomized" not "customized" (S missing)

### Zigbee2MQTT version

2.4.0

### Adapter firmware version

Revision: 0x26580700

### Adapter

ConBee2/RaspBee2 0x00212effff074f0e Revision: 0x26580700

### Setup

Container within Unraid.

Koenkk, 2025-07-21:
Could you check if this is fixed in 2.5.1? (you will need to change the color power on behaviour once after updating0

FransHalsjes, 2025-07-22:
Upgraded to 2.5.1, changed power on behavior in Z2MQTT frontend. Restarted Home Assistant. 
Problem gone. Thanks for the quick response!