Koenkk/zigbee2mqtt #28042: Errors zh:controller:greenpower: [FULLENCR] srcID=3791806893 gpp=NO commandIdentifier=0 Unknown security key

dbizien, 2025-07-21:
### What happened?

Hi,
I have re-installed from scratch the solution and I keep to have some errors every seconds.
I guess there is a device that is trying to comunicate.
I have no problem with devices. I just want to have less error messages.

What I have done : 
- I have install and desinstall device without error except those I receive every seconds.
- I am in the latest version. I have tried with several old ones and dev latest => same error
- I am using SLZB-06M, I have tried to change the zigbee version or core version => same error

[log.log](https://github.com/user-attachments/files/21348598/log.log)

 I have attached the log in debug mode and info mode.

There is no device in the database.

### What did you expect to happen?

I expect to have no errrors every seconds.
It seems there is also a crash : 
error: 	zh:controller:greenpower: TypeError: Cannot read properties of undefined (reading 'options')
    at GreenPower.processCommand (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/greenPower.ts:347:64)

But if I can have a workaround to exclude those devices, it will be ok for me.

### How to reproduce it (minimal and precise)

I think the problem is specific to my config as I have never seen that elsewhere.

### Zigbee2MQTT version

    2.5.1 commit: 5063c7d3ad88f9ed651dc36fd9e583ff452e18da 

### Adapter firmware version

    8.0.2 [GA]

### Adapter

SLZB-06M

### Setup

Running Z2M from a synology NAS via conainer manager and a docker file.

### Device `database.db` entry

{ 	"id": 1, 	"type": "Coordinator", 	"ieeeAddr": "0x84b4dbfffec7a9be", 	"nwkAddr": 0, 	"manufId": 4169, 	"epList": [ 		1, 		242 	], 	"endpoints": { 		"1": { 			"profId": 260, 			"epId": 1, 			"devId": 101, 			"inClusterList": [ 				0, 				3, 				6, 				8, 				10, 				25, 				768 			], 			"outClusterList": [ 				0, 				3, 				4, 				5, 				6, 				8, 				32, 				768, 				1024, 				1026, 				1029, 				1030, 				1280, 				1794, 				2817, 				2819, 				2820, 				4096 			], 			"clusters": {}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		}, 		"242": { 			"profId": 41440, 			"epId": 242, 			"devId": 102, 			"inClusterList": [ 				33 			], 			"outClusterList": [ 				33 			], 			"clusters": {}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"interviewCompleted": true, 	"interviewState": "SUCCESSFUL", 	"meta": {} }

### Debug log

[log.log](https://github.com/user-attachments/files/21348819/log.log)

Nerivec, 2025-07-22:
Something appears to be very wrong with your adapter.
Since you have a brand new network, can you try clearing the NVM3 on your adapter, to make sure it's factory-clean?
https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3

dbizien, 2025-07-22:
Hi,
I'm using a Synology NAS with Container Manager and docke-compose.yml.
I keep on my file system just the configuration file.
It is easy to delete everything and recreate it.
That is why I'm sure it is factory clean ...
That is why also the database.db is clean because I have deleted it.

But, I agree, there is something strange.

I even try to add 4 devices (no problem) and to put their id in the passlist.
I keep receiving errors every seconds ....

Nerivec, 2025-07-22:
The NVM3 clear is for the adapter.

dbizien, 2025-07-22:
Hi,
I have not succeed the solution with NVM3 clear.
I tried also to flash the adapter with other config matter, zigbee router to clean it but when I'm back at coordinator I have the same problem.
Do you think there is a way to block those errors ?

Nerivec, 2025-07-23:
> I have not succeed the solution with NVM3 clear.

Do you mean it didn't change anything after clearing it, or you were not able to clear it?

Do you have any GreenPower devices around you that could cause the adapter to receive these frames?

dbizien, 2025-07-23:
Hi,
I have flashed the SLZB-06M and I still have the problem.
I will try to remove some powertag to try to check if can come from there.

dbizien, 2025-07-26:
Hi,
I have disconnected most of the Powertag and I have no more errors.
The problem was coming from those devices.
I will try to reconnect them seems there is a solution with an ikea device ...
Thanks a lot !

closed 2025-07-26 by dbizien