Koenkk/zigbee2mqtt #28052: Missing Exposes for Backlight, Power-On Behavior on TS0003 Tuya 3 Gang Switch

allanpocostaofc, 2025-07-22:
### What happened?

I have some Tuya `TS0003` 3-gang switches on my Zigbee2MQTT. They are being exposed correctly, but the `Backlight` and `Power-On Behavior` attributes are not being displayed for this model.

[image]

### What did you expect to happen?

That both attributes were exposed in `Zigbee2MQTT`.

### Zigbee2MQTT version

2.5.1-1

### Adapter firmware version

8.0.2 [GA]

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Add-on on Home Assistant OS on Intel N100 with Linux.
Linux proxmox 6.8.12-4-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-4 (2024-11-06T15:04Z) x86_64 GNU/Linux
Debian GNU/Linux 12

### Device `database.db` entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c138eb6429a4bd","nwkAddr":52017,"manufId":4417,"manufName":"_TZ3000_66fekqhh","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genBasic":{"attributes":{"65503":"\u0019\u0012\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_66fekqhh","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0ceff6fffecdcc57","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0ceff6fffecdcc57","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x0ceff6fffecdcc57","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1753210674730}

bernardesarthur, 2025-08-02:
I fix that device here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/9575

Btw I have that device and the exposes works fine:

[image]

[image]

allanpocostaofc, 2025-08-02:
In today's update, my devices received the properties correctly. 

[image]

closed 2025-08-11 by allanpocostaofc