Koenkk/zigbee2mqtt #28064: [Wrong device]: MOES ZK-EU shows as TS011F_plug_1

gontxomde, 2025-07-24:
### Link

https://www.aliexpress.com/item/1005004556206576.html

### Model

ZK-EU

### Description

Smart wallsocket that also monitors power usage

### Vendor

Moes

### Picture (link)

https://ae01.alicdn.com/kf/Sce4929a9aabd4a1fa4a8fcd080ae87a4G/Moes-ZigBee-Wall-Socket-WIFI-Smart-Glass-Panel-Outlet-Power-Monitor-tuya-Wireless-Control-Timer-Alexa.jpg_640x640.jpg

### Database entry

I dont know how to retrieve it in HA

### Notes

The Energy, Voltage, Power and current are not being reported. I don't know how to retrieve the database.db information. if someone could guide me I will add it ASAP

Koenkk, 2025-07-24:
Could you provide the debug log when configuring the device through the device -> frontend -> about -> refresh icon?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

gontxomde, 2025-07-24:
## Logs in the HA-ADDOn Menu
[2025-07-24 20:43:17] debug: 	zh:controller:endpoint: ZCL command 0xa4c1386b2b6341b0/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-07-24 20:43:17] debug: 	zh:deconz: ZCL request sent with transactionSequenceNumber.: 24
[2025-07-24 20:43:17] debug: 	zh:deconz: command.response: 1, zcl.disableDefaultResponse: true, z2m.disableResponse: false, request.timeout: 10000
[2025-07-24 20:43:17] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x267 EP:1 seq: 42 requestId: 236
[2025-07-24 20:43:17] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 236 status: Success
[2025-07-24 20:43:17] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 44
[2025-07-24 20:43:17] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x0267 APS request id: 236 confirm status: Success 0x00
[2025-07-24 20:43:17] debug: 	zh:deconz:driver: Request APS-DATA.indication seq: 45
[2025-07-24 20:43:17] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 45 srcAddr: 0x267 destAddr: 0x0 profile id: 0x0104 cluster id: 0x0000 lqi: 103
[2025-07-24 20:43:17] debug: 	zh:controller: Received payload: clusterID=0, address=615, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=103, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":24,"commandIdentifier":1},"payload":[{"attrId":4,"status":0,"dataType":66,"attrData":"_TZ3000_o1jzcxou"},{"attrId":0,"status":0,"dataType":32,"attrData":3},{"attrId":1,"status":0,"dataType":32,"attrData":192},{"attrId":5,"status":0,"dataType":66,"attrData":"TS011F"},{"attrId":7,"status":0,"dataType":48,"attrData":1},{"attrId":65534,"status":0,"dataType":48,"attrData":0}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-07-24 20:43:17] debug: 	zh:deconz: Response received transactionSequenceNumber: 24
[2025-07-24 20:43:17] debug: 	z2m: Received Zigbee message from 'enchufe_pared_gonzalo_1', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":192,"manufacturerName":"_TZ3000_o1jzcxou","modelId":"TS011F","powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0
[2025-07-24 20:43:17] debug: 	zh:controller:endpoint: Bind 0xa4c1386b2b6341b0/1 genOnOff from '0x00212effff0771c1/1'
[2025-07-24 20:43:17] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x267 EP:0 seq: 46 requestId: 237
[2025-07-24 20:43:17] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 237 status: Success
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 48
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x0267 APS request id: 237 confirm status: Success 0x00
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.indication seq: 49
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 49 srcAddr: 0x267 destAddr: 0x0 profile id: 0x0000 cluster id: 0x8021 lqi: 103
[2025-07-24 20:43:18] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=SUCCESS, payload=undefined
[2025-07-24 20:43:18] debug: 	zh:controller:endpoint: Bind 0xa4c1386b2b6341b0/1 haElectricalMeasurement from '0x00212effff0771c1/1'
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x267 EP:0 seq: 50 requestId: 238
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 238 status: Success
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 52
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x0267 APS request id: 238 confirm status: Success 0x00
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.indication seq: 53
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 53 srcAddr: 0x267 destAddr: 0x0 profile id: 0x0000 cluster id: 0x8021 lqi: 103
[2025-07-24 20:43:18] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=SUCCESS, payload=undefined
[2025-07-24 20:43:18] debug: 	zh:controller:endpoint: Bind 0xa4c1386b2b6341b0/1 seMetering from '0x00212effff0771c1/1'
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.request: dest: 0x267 EP:0 seq: 54 requestId: 239
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: Response APS-DATA.request APS request id: 239 status: Success
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.confirm seq: 56
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: APS-DATA.confirm  destAddr: 0x0267 APS request id: 239 confirm status: Success 0x00
[2025-07-24 20:43:18] debug: 	zh:deconz:driver: Request APS-DATA.indication seq: 57
[2025-07-24 20:43:18] debug: 	zh:deconz:frameparser: Response APS-DATA.indication seq: 57 srcAddr: 0x267 destAddr: 0x0 profile id: 0x0000 cluster id: 0x8021 lqi: 119
[2025-07-24 20:43:18] debug: 	zh:controller: Received ZDO response: clusterId=BIND_RESPONSE, status=SUCCESS, payload=undefined
[... cut ...]
info 2025-07-24 20:44:46z2m:mqtt: MQTT publish: topic 'izaguirre/bridge/response/device/configure', payload '{"data":{"id":"enchufe_pared_gonzalo_1"},"status":"ok","transaction":"uddpq-1"}'
info 2025-07-24 20:44:53z2m: Interviewing 'enchufe_pared_gonzalo_1'
info 2025-07-24 20:44:53zh:controller:device: Device '0xa4c1386b2b6341b0' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
info 2025-07-24 20:44:55z2m: Successfully interviewed 'enchufe_pared_gonzalo_1'
info 2025-07-24 20:44:55z2m:mqtt: MQTT publish: topic 'izaguirre/bridge/response/device/interview', payload '{"data":{"id":"enchufe_pared_gonzalo_1"},"status":"ok","transaction":"uddpq-2"}'

Koenkk, 2025-07-25:
It seems that your plug doesn't support reporting so we need to poll the values, can you:
- stop z2m
- in the `database.db` change two times `"appVersion":192` to `"appVersion":160`
- start z2m
- the plug should now be detected as `TS011F_plug_3`
- see if values are now reported.