Koenkk/zigbee2mqtt #28066: ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

LanMorehell, 2025-07-24:
### What happened?

Good day!

1. HA on Pi
2. Z2m starts and works just fine for ~12h
3. Adapter crashes and disconnects with te following errors:

`[2025-07-24 08:23:00] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-07-24 08:23:00] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-07-24 08:23:00] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-07-24 08:23:00] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-07-24 08:23:00] debug: 	zh:controller: Adapter disconnected`

Restart of the HA resets the process for another 2hrs. Log attached
Initially happened with Sonoff dongle E. I thought it was an adapter problem, so transferred to SMLIGHT SLZB-07MG24. Same stuff happens.

### What did you expect to happen?

I expected Z2m to work continously without interruptions

### How to reproduce it (minimal and precise)

1. restart HA
2. use automations
3. Wait 12 hrs

### Zigbee2MQTT version

 2.5.1 commit: unknown

### Adapter firmware version

 7.4.4 [GA]

### Adapter

Smlight slzb-07mg24, sonoff dongle-e

### Setup

HA add-on, HA running on Pi

Config:
`devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqttacc
  password: Solomon
  server: mqtt://core-mosquitto:1883
serial:
  adapter: ember
  port: >-
    /dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-07Mg24_58c12ec323adef11b8608a4ba8793231-if00-port0
  baudrate: 115200
  rtscts: true
advanced:
  log_level: debug
  pan_id: 6753
  channel: 15
  network_key:
    - 36
    - 109
    - 207
    - 36
    - 213
    - 187
    - 141
    - 206
    - 246
    - 115
    - 170
    - 205
    - 10
    - 80
    - 155
    - 170
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  enabled: true
  port: 8099
availability:
  enabled: false
version: 4
ota:
  image_block_response_delay: 200
  default_maximum_data_size: 15`

### Debug log

[log.log](https://github.com/user-attachments/files/21404490/log.log)

Nerivec, 2025-07-25:
Can you confirm this happens with latest [8.0.2 firmware](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update7)?
8.2.x is too new, not a lot of feedback on it.

ParalaXEngineering, 2025-08-28:
Hi,

So a bit of news. Everything was all and well for about 5 days, then I had a restart from zigbee2mqtt. But this time, it was different: the zigbee chip itself has restarted, but not the SLZB. I don't see any log on the device that mention what happened, and here is what zigbee2mqtt tells me:
[2025-08-27 12:53:29] error: 	zh:ember:uart:ash: Port Error: read ECONNRESET
[2025-08-27 12:53:29] error: 	zh:ember: Adapter fatal error: ERROR_SERIAL_INIT
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Total frames: RX=512, TX=572
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   DATA frames : RX=512, TX=60
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   DATA bytes  : RX=16692, TX=1695
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=512
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-08-27 12:53:29] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-08-27 12:53:29] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-08-27 12:53:29] error: 	z2m: Adapter disconnected, stopping
[2025-08-27 12:53:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-08-27 12:53:29] info: 	z2m: Disconnecting from MQTT server
[2025-08-27 12:53:29] info: 	zh:ember:uart:ash: Port closed.
[2025-08-27 12:53:29] info: 	z2m: Stopping zigbee-herdsman...
[2025-08-27 12:53:29] info: 	z2m: Stopped zigbee-herdsman
[2025-08-27 12:53:29] info: 	z2m: Stopped Zigbee2MQTT

Nerivec, 2025-08-28:
`ECONNRESET` means the slzb could no longer be reached (network connection lost). This can happens for various reasons, network stability issues (router overloaded/updated/restarted/etc.), slzb under too much stress, (possibly z2m machine under too much stress)...

ParalaXEngineering, 2025-09-06:
Hi,

Sorry I missed your answer. But in any case, it's been a while now, and no error, so it is definitely on the right track! Thanks a lot for your help, I wouldn't have figured out...