Koenkk/zigbee2mqtt #28068: [New device support]: MOES Tuya Zigbee Garage Door Controller Module ZM-102-M

ka62l, 2025-07-24:
### Link

https://www.aliexpress.com/item/1005009332309392.html?spm=a2g0o.order_list.order_list_main.35.1b0c5e5b6hHEM2&gatewayAdapt=4itemAdapt

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xa4c1386f74948f0d","nwkAddr":39047,"manufId":4417,"manufName":"_TZE204_jktmrpoj","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1753363471608}

### Zigbee2MQTT version

2.5.1

### Comments

Hello, I try to add a tuya zigbee Remote Garage Door Openerand it is not compatible with Zigbee2MQTT.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_jktmrpoj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

All the fonction doesn't work

bernardesarthur, 2025-07-28:
I have a similar device so we can try this converter to see if that works.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const m = require('zigbee-herdsman-converters/lib/modernExtend');
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const e = exposes.presets;
const ea = exposes.access;

```
const definition = {
    {
        fingerprint: tuya.fingerprint("TS0601", ["_TZE204_nklqjk62", "_TZE200_nklqjk62"]),
        model: "PJ-ZGD01",
        vendor: "Tuya",
        description: "Garage door opener",
        whiteLabel: [{vendor: "MatSee Plus", model: "PJ-ZGD01"}],
        extend: [
            tuyaMagicPacket(),
            dpBinary({
                name: "trigger",
                dp: 1,
                type: tuya.dataTypes.bool,
                valueOn: [true, true],
                valueOff: [false, false],
                description:
                    "Request door to close (= false) or open (= true), will not pulse output if contact shows door is already in requested state",
            }),
            dpNumeric({
                name: "countdown",
                dp: 2,
                type: tuya.dataTypes.number,
                description: "Countdown to trigger door movement after a certain time, will pulse output in all cases",
                unit: "s",
                valueMin: 0,
                valueMax: 43200,
            }),
            dpBinary({
                name: "garage_door_contact",
                dp: 3,
                type: tuya.dataTypes.bool,
                valueOn: [true, false],
                valueOff: [false, true],
                description: "Indicates if the garage door contact is closed (= true) or open (= false)",
                readOnly: true,
            }),
            dpNumeric({
                name: "run_time",
                dp: 4,
                type: tuya.dataTypes.number,
                description: "Configure the time to wait for the door contact status to change before triggering a run time alarm",
                unit: "s",
                valueMin: 0,
                valueMax: 120,
            }),
            dpNumeric({
                name: "open_alarm_time",
                dp: 5,
                type: tuya.dataTypes.number,
                description: "Configure the amount of time the door may be open before an open time alarm is triggered",
                unit: "s",
                valueMin: 0,
                valueMax: 86400,
            }),
            dpEnumLookup({
                name: "status",
                dp: 12,
                type: tuya.dataTypes.enum,
                description: "Indicates run time alarm, door open alarm or normal status, will not return to normal until door is triggered again",
                lookup: {"Open Time Alarm": 0, "Run Time Alarm": 1, Normal: 2},
                readOnly: true,
            }),
        ],
  };

module.exports = definition;
```

HomonayV, 2025-07-28:
> > Hello, looking for the same device "MOES Tuya Zigbee Garage Door Controller Module ZM-102-M" support in Zigbee2MQTT. Also bought from Aliexpress "MOES Official Store". Thanks in advance!
> 
> Can you provide your external definition?

export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_jktmrpoj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

HomonayV, 2025-07-28:
And when deleted Unexpected token '{'

[2025-07-29 00:18:12] debug: 	z2m: TypeError: tuyaMagicPacket is not a function
    at Object.<anonymous> (/config/zigbee2mqtt/external_converters/.tmp-ed42d4f2-moes-a6be6fff-04df-4f87-a3cc-9c64ee793f43.js:20:9)
    at Module._compile (node:internal/modules/cjs/loader:1730:14)
    at Object..js (node:internal/modules/cjs/loader:1895:10)
    at Module.load (node:internal/modules/cjs/loader:1465:32)
    at Function._load (node:internal/modules/cjs/loader:1282:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
    at cjsLoader (node:internal/modules/esm/translators:266:5)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:200:7)
    at ModuleJob.run (node:internal/modules/esm/module_job:274:25)

bernardesarthur, 2025-07-28:
> > I have a similar device so we can try this converter to see if that works.
> > ```
> > const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> > const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> > const exposes = require('zigbee-herdsman-converters/lib/exposes');
> > const legacy = require('zigbee-herdsman-converters/lib/legacy');
> > const tuya = require('zigbee-herdsman-converters/lib/tuya');
> > const m = require('zigbee-herdsman-converters/lib/modernExtend');
> > const reporting = require("zigbee-herdsman-converters/lib/reporting");
> > const e = exposes.presets;
> > const ea = exposes.access;
> > 
> > const definition = {
> >     {
> >         fingerprint: tuya.fingerprint("TS0601", ["_TZE204_jktmrpoj"]),
> >         model: "PJ-ZGD01",
> >         vendor: "Tuya",
> >         description: "Garage door opener",
> >         extend: [
> >             tuyaMagicPacket(),
> >             dpBinary({
> >                 name: "trigger",
> >                 dp: 1,
> >                 type: tuya.dataTypes.bool,
> >                 valueOn: [true, true],
> >                 valueOff: [false, false],
> >                 description:
> >                     "Request door to close (= false) or open (= true), will not pulse output if contact shows door is already in requested state",
> >             }),
> >             dpNumeric({
> >                 name: "countdown",
> >                 dp: 2,
> >                 type: tuya.dataTypes.number,
> >                 description: "Countdown to trigger door movement after a certain time, will pulse output in all cases",
> >                 unit: "s",
> >                 valueMin: 0,
> >                 valueMax: 43200,
> >             }),
> >             dpBinary({
> >                 name: "garage_door_contact",
> >                 dp: 3,
> >                 type: tuya.dataTypes.bool,
> >                 valueOn: [true, false],
> >                 valueOff: [false, true],
> >                 description: "Indicates if the garage door contact is closed (= true) or open (= false)",
> >                 readOnly: true,
> >             }),
> >             dpNumeric({
> >                 name: "run_time",
> >                 dp: 4,
> >                 type: tuya.dataTypes.number,
> >                 description: "Configure the time to wait for the door contact status to change before triggering a run time alarm",
> >                 unit: "s",
> >                 valueMin: 0,
> >                 valueMax: 120,
> >             }),
> >             dpNumeric({
> >                 name: "open_alarm_time",
> >                 dp: 5,
> >                 type: tuya.dataTypes.number,
> >                 description: "Configure the amount of time the door may be open before an open time alarm is triggered",
> >                 unit: "s",
> >                 valueMin: 0,
> >                 valueMax: 86400,
> >             }),
> >             dpEnumLookup({
> >                 name: "status",
> >                 dp: 12,
> >                 type: tuya.dataTypes.enum,
> >                 description: "Indicates run time alarm, door open alarm or normal status, will not return to normal until door is triggered again",
[... 64 lines cut ...]
                valueMin: 0,
                valueMax: 120,
            }),
            dpNumeric({
                name: "open_alarm_time",
                dp: 5,
                type: tuya.dataTypes.number,
                description: "Configure the amount of time the door may be open before an open time alarm is triggered",
                unit: "s",
                valueMin: 0,
                valueMax: 86400,
            }),
            dpEnumLookup({
                name: "status",
                dp: 12,
                type: tuya.dataTypes.enum,
                description: "Indicates run time alarm, door open alarm or normal status, will not return to normal until door is triggered again",
                lookup: {"Open Time Alarm": 0, "Run Time Alarm": 1, Normal: 2},
                readOnly: true,
            }),
        ],
  };

module.exports = definition;
```

bernardesarthur, 2025-07-28:
> export default { zigbeeModel: ['TS0601'], model: 'TS0601', vendor: '_TZE204_jktmrpoj', description: 'Automatically generated definition', extend: [], meta: {}, };

Try the same converter just replace your manufacturer to yours **_TZE204_jktmrpoj**.

bernardesarthur, 2025-07-29:
Try this one both of you:

Could you try to:

1. Stop Z2M
**2. Open data/database.db with a text editor and change all occurrences of _TZE204_jktmrpoj  to _TZE204_nklqjk62**
3. Start Z2M and see if the device now works

Let me know if that works.

ka62l, 2025-07-29:
> Try this one both of you:
> 
> Could you try to:
> 
> 1. Stop Z2M
> 2. Open data/database.db with a text editor and change all occurrences of **_TZE204_jktmrpoj  to _TZE204_nklqjk62**
> 3. Start Z2M and see if the device now works
> 
> Let me know if that works.

It's good, the device work fine.
Best regards.

closed 2025-07-29 by Koenkk

HomonayV, 2025-07-29:
> Good bro! I'll add your manufacturer in the next release. Thanks

Thank bro! In my case working too!

seler, 2025-08-19:
I've got a problem with those devices. They spam my network several times a second. Anyone else experiencing this? Tried 2 so far and both have the same issue.

onlinehunter, 2025-11-21:
I also have the same problem with this MOES device
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_jktmrpoj',
    description: 'Automatically generated definition',
    extend: [],
};

Zigbee2MQTT version 2.6.3
commit: 6e001b48ec278c6c141579c9eacc045f190fc289
Frontend version 2.2.3
zigbee-herdsman-converters version 25.59.0
zigbee-herdsman version 6.3.2
Adapter EZSP v8
Setup os: `#88~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Oct 14 14:03:14 UTC 2 - 6.8.0-87-generic - x64`
node: `v22.16.0`
ha: `true`

[image]

mairhart, 2025-12-04:
Same issue with constant spamming of the ZigBee network from the device.

> I've got a problem with those devices. They spam my network several times a second. Anyone else experiencing this? Tried 2 so far and both have the same issue.
> 
> <img alt="Image" width="1367" height="747" src="https://private-user-images.githubusercontent.com/255675/479506708-ed3bd2d6-05e9-41e2-b3cf-cfb5b7229ce8.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ4MjU2MTMsIm5iZiI6MTc2NDgyNTMxMywicGF0aCI6Ii8yNTU2NzUvNDc5NTA2NzA4LWVkM2JkMmQ2LTA1ZTktNDFlMi1iM2NmLWNmYjViNzIyOWNlOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTIwNFQwNTE1MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZDk0NTI1MGFlYWNhZjM3MzQ4ZjZlNDVkYzMwMGY0YjQ1YjE1YTQ3OWZjNjJmOTVmYmZkY2Y5NzAyYWIxYWJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.FxR9leXTVK-ae88nww-W_CgeactIt89KE7JzKCayHSQ">