Koenkk/zigbee2mqtt #28073: Hue 9290012573A (LCT015) not checking in.

ExquisiteMetropolis, 2025-07-25:
### What happened?

I have several Hue E27 color lights.
Two are of LCT015 / 9290012573A type. 
Hue white and color ambiance E26/E27/E14

One of them is on firmware ID:
Firmware build date: 20240329
Firmware ID: 67.116.3

The other on:
Firmware build date: 20240329
Firmware ID: 1.116.3

Looks somewhat similar, not fully. Not sure why.

Other E27 color lights I have are of a slightly different models: 
LCA001/ 9290022166
LCA006 / 9290024688
Running on this firmware:
Firmware build date: 20250213
Firmware ID: 1.122.8

The issue I have also is that the LCT015 lamps are not regularly checking in with the controller. Last time was over 1 week already. Unless I power-cycle them. (Didn't have the need to turn on these two specific lights for a while now).
My 'offline zigbee device alert' doesn't work properly, as these lights always appear to be offline. And I'd like to get rid of that fact.

As far as I have been able to determine, the LCT015 lights are WITHOUT bluetooth.
That might explain the difference in firmware. Can anyone confirm 1.116.3 / 67.116.3 is the latest version for these lights? https://www.philips-hue.com/en-us/support/release-notes/lamps does not show anything beyond OCT-2024.

Then the question remains, how can I ensure that the LCT015 check in with the controller every 1 hour or so, like all other hue lights do. 

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet Sonoff-E

### Setup

HA OS on Proxmox VM, Z2M as HA Addon

TheJulianJES, 2025-07-26:
1.116.3/67.116.3 are indeed the latest versions for the older Atmel (and for TI based) Hue lights. They run a way older Zigbee stack than the newer Silabs based Hue lights, but they should still work fine for the most part.
However, they don't support some features like attribute reporting, meaning they have to be polled if you want to get their current state. Since they're much less chatty on their own compared to newer bulbs, they might inadvertently trigger Z2M's offline device detection.

I'm not sure how that detection works in Z2M, but assuming it doesn't try to (e.g.) read an attribute from the Hue light after some time of inactivity to confirm a node is dead, you might see the node being marked offline until there's another message.

The Hue Bridge itself polls the lights every couple of seconds, depending on how many you have connected. So, I guess you might be able to use `homeassistant.update_entity` or some MQTT calls to just read an attribute on the older Hue lights every few hours yourself. That *should* prevent Z2M marking them as offline, as the Hue lights return a "read attribute response".

ExquisiteMetropolis, 2025-07-27:
@Nerivec & @TheJulianJES thank you for taking the time to reply to my query. 

I think I didn't explain myself correctly, the device doesn't show offline in Z2M. It just shows as this:

LCT015:
[image]

Where the other lights do check-in more frequently (LCA001 example:)

[image]

The offline reference I made, is a script / automation I have running within Home Assistant. 
This checks if a device hasn't been seen for +1 hour. (Usefull when visitors flick a light switch for example, prevents me from wondering why lights don't function anymore 'all of the sudden'). 

The two LCT015: lights in question:
```
  '0x0017880102xxxxxx':
    friendly_name: Light 1
    transition: 2

  '0x0017880104xxxxxx':
    friendly_name: Light 2
    transition: 2
```

Example for LCA001:

```
  '0x0017880106xxxxxx':
    friendly_name: Light 6
    transition: 2

```

I also checked the config file, I don't have "availability" enabled in general.  
it's also not specified for any of the lights. Not sure why I never turned this on. Might fix my issue. I will look into this! Interesting. 👍 
Although I do would expect a light to report in from time to time regardless of this setting? (correct or incorrect assumption?)

ExquisiteMetropolis, 2025-07-27:
Seems to be working now, with the availability enabled. Thanks for the push in the right direction. 👍

closed 2025-07-27 by ExquisiteMetropolis