Koenkk/zigbee2mqtt #28088: Bosch Siren Outdoor (BSIR-EZ) impossible to pair.

domoarto, 2025-07-27:
### What happened?

Just purchased one BSIR-EZ and as already reported on some closed topics (with no solution), it's currently impossible to pair BSIR-EZ (https://www.zigbee2mqtt.io/devices/BSIR-EZ.html) with Zigbee2MQTT.
Install code have been added to Z2M before trying to pair.

For info and as mentionned in link above : "Instead, scan the QR code with your phone and this will return a 91 long code"
I'm not the only one but my current Install code isn't 91 long code but 95. I don't know if it can be one of the roots of the current issue.

When trying to pair, Z2M is able to detect the Bosch Alarm but for some milliseconds ... followed by "Left the network"
And then it's looping like this for ever like every 8-10 seconds.

Here are the logs lines from Z2M : (2 times here)

`info 2025-07-27 13:12:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x287681fffe28166e"},"type":"device_removed"}'
info 2025-07-27 13:12:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e"},"type":"device_leave"}'
info 2025-07-27 13:14:01zh:controller: Interview for '0x287681fffe28166e' started
info 2025-07-27 13:14:01z2m: Device '0x287681fffe28166e' joined
info 2025-07-27 13:14:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x287681fffe28166e"},"type":"device_connected"}'
info 2025-07-27 13:14:01z2m: Starting interview of '0x287681fffe28166e'
info 2025-07-27 13:14:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x287681fffe28166e"},"type":"pairing"}'
info 2025-07-27 13:14:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e"},"type":"device_joined"}'
info 2025-07-27 13:14:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e","status":"started"},"type":"device_interview"}'
warning 2025-07-27 13:14:02z2m: Device '0x287681fffe28166e' left the network
info 2025-07-27 13:14:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x287681fffe28166e"},"type":"device_removed"}'
info 2025-07-27 13:14:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e"},"type":"device_leave"}'
info 2025-07-27 13:14:09zh:controller: Interview for '0x287681fffe28166e' started
info 2025-07-27 13:14:09z2m: Device '0x287681fffe28166e' joined
info 2025-07-27 13:14:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x287681fffe28166e"},"type":"device_connected"}'
info 2025-07-27 13:14:09z2m: Starting interview of '0x287681fffe28166e'
info 2025-07-27 13:14:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x287681fffe28166e"},"type":"pairing"}'
info 2025-07-27 13:14:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e"},"type":"device_joined"}'
info 2025-07-27 13:14:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x287681fffe28166e","ieee_address":"0x287681fffe28166e","status":"started"},"type":"device_interview"}'
warning 2025-07-27 13:14:09z2m: Device '0x287681fffe28166e' left the network`

Tested with Z2M 2.5.1, also tried with 1.42.0 just for testing, same behavior.

[image]
[image]

### How to reproduce it (minimal and precise)

Try to pair a BSIR-EZ ...

### Zigbee2MQTT version

2.5.1 // 1.42.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet / Sonoff ZBDongle-E (using ember adapter type in configuration)

### Setup

Pi5, full Dockerized environnement, Z2M aswell.

domoarto, 2025-07-30:
Hi,
If there is any ways on which i can provide help, do not hesitate to tell me.
Thanks a lot in advance !

smeegoan, 2025-08-21:
I also have the same problem and my Bosch siren install code also has 95 chars. Attached are the debug logs. They are a bit large because I have a lot of zigbee devices

[45df7312_zigbee2mqtt_2025-08-21T21-01-02.612Z.log](https://github.com/user-attachments/files/21926280/45df7312_zigbee2mqtt_2025-08-21T21-01-02.612Z.log)

frankforpresident, 2025-08-31:
Same issue here, also the 95 chars. Just order from amazon and brand new

Nerivec, 2025-09-01:
You have to look for the key exchange (should start right after association is done). That's likely what's not properly happening (hence device leaves). _As mentioned before, some Bosch stuff uses an old Zigbee 1.2 spec, and these are very problematic with 3.0 since they don't follow proper security measures & whatnot. If that's the case with these alternative models, there's simply nothing we can do to fix it (they were made to work with the Bosch hub basically... nothing else works on this older spec anymore)._

Feel free to use my set of coloring rules for Wireshark (View>Coloring rules>Import), should make it easier to identify important Zigbee stuff: [coloring-rules.txt](https://github.com/user-attachments/files/22084257/coloring-rules.txt)
Otherwise, you can send over the file, here or privately if you prefer (can find me on Discord, or setup a private repo on GitHub...).

frankforpresident, 2025-09-02:
I do see an association request and a response `Association Status: 0x00 (Association Successful)` but I don't see any key exchange. I will share the files via discord just to confirm.

Is there a way see if the device is using the 1.2 spec instead of 3.0? In the spec sheet from the manual they claim the protocol is _zigbee 3.0_. 

Is it worth it to buy a bosch hub to see how they communicate with the sniffer? I still have until the end of september, otherwise I need to return the siren back to amazon.

Some extra information:
I used a Sonoff Dongle-P plus (zstack) with firmware 20240710 and a CC2531 usb stick flashed with ZBOSS for the sniffing.

frankforpresident, 2025-09-02:
Alright, I found the issue 🤦 ... and it is stupid. 
 
All the dip switches on my _new_ unit where set to the OFF position while according to the documentation the factory settings should be ON. After toggling DIP 8 to the ON position the siren was recognised and successfully paired.

Maybe we can use this ticket to update the notes in the [documentation](https://www.zigbee2mqtt.io/devices/BSIR-EZ.html) regarding the [factory DIP settings
](https://tt-smarthome.resource.bosch.com/nrt/web/produkt/aussensirene/pdf/8-750-001-502_V002_Outdoor_siren_QSG_EN_20210730_web.pdf)

frankforpresident, 2025-09-02:
In the end I've set all the switches to the ON position but DIN 8 is the most important one as this one enables "smart" mode. it would have been nice that the siren was not discoverable at all when the switch is set to OFF but it is what it is..

I will update the documentation in the evening..

smeegoan, 2025-09-02:
I can also confirm with the switch 8 in the ON position it works 👍

nachonam, 2025-09-03:
Hum! I think perhaps there's a problem with the factory settings. 
In my case all the switches were UP and I had to change the DIP 8 to down

nachonam, 2025-09-03:
Yes the Siren is running… but the doc must be updated..

IMPORTANT: https://github.com/Koenkk/zigbee2mqtt/issues/28088 factory settings and set switch number 8 to OFF - "Legacy" mode

frankforpresident, 2025-09-03:
I disagree, the docs should be written more generic because for me it needed to be set to "ON" aka "smart" mode as mine was by default on OFF..  Also we have some reports from users like @smeegoan reporting "ON" worked for them

What do you think of:

IMPORTANT: [Check DIP switchers](https://github.com/Koenkk/zigbee2mqtt/issues/28088).  If authorization is failing and your device is leaving the network, it might help to toggle DIP "8" to the ON or OFF position, depending on how the unit was initially shipped.

nachonam, 2025-09-03:
Hmm! There may be a difference with the factory default settings depending on the point of sale (for me, France via Amazon FR).
In my case, all the switches were ON (up) on both devices, and I switched DIP 8 to OFF (down).

smeegoan, 2025-09-03:
In my case I bought from Amazon ES and all the switches were also down. The only way I got the Siren to pair was when i moved the DIP 8 up. Here is the photo of my DIP switches:

[image]