Koenkk/zigbee2mqtt #28094: IKEA INSPELNING smart plug not joining

thomasreiser, 2025-07-28:
### What happened?

I've been trying for days to pair my new IKEA INSPELNING outlet with Zigbee2MQTT. I enter pairing mode by pressing the small pairing button for 3 (or 5) seconds. The LED pulses briefly – and that's it.

I also tried to place the INSPELNING some centimeters near the Zigbee stick without any success.

In Zigbee2MQTT, I don't see any logs – nothing at all.

### What did you expect to happen?

Seeing at least any logs in Zigbee2MQTT.

### How to reproduce it (minimal and precise)

In my setup:

- Enable pairing in Z2M
- Plug INSPELNING into outlet
- Bring INSPELNING in pairing mode
- Nothing happens

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

0x26510900

### Adapter

ConBee3

### Setup

Add-on on Home Assistant OS

closed 2025-08-01 by Nerivec

thomasreiser, 2025-08-08:
@Koenkk @Nerivec sorry I was not at home but now able to try it with the latest 2.6.0 release - but without success.

Will try to update the Conbee3 this weekend.

dicotr, 2025-08-09:
Okay - Firmware update of the ConBee3 to the latest 0x26550900 fixed it. Thank you for all your efforts.

thomasreiser, 2025-08-09:
Okay - firmware update of the ConBee3 to the lastest 0x26550900 fixed it. Thank you for all your efforts.