Koenkk/zigbee2mqtt #28099: Request for device support (ready to paste) markdown - Model: TS0601 - Manufacturer: _TZE284_ye5jkfsb

vuvan89cz, 2025-07-29:
### Device model information
- Model: TS0601
- Manufacturer: _TZE284_ye5jkfsb
- Description: Unknown Tuya-based device (not currently supported)

### Exposes
The device currently shows as "Not supported" and no exposes are visible in the web interface.

### Additional information
Please add support for this Tuya device. Let me know if you need any additional logs or testing. I'm willing to help with integration.

### Similar supported device
There is a similar supported device from the same manufacturer:
- Model: TS0601
- Manufacturer: _TZE200_ye5jkfsb
- Product: Moes BHT-002 Thermostat
- Zigbee2MQTT reports it as supported and functional.

It seems my device (_TZE284_ye5jkfsb) may be a variant or updated version of the same Moes BHT series thermostat.

### Exposes
Currently, no exposes are visible in Zigbee2MQTT. I can assist with testing or collecting more data.

### Additional information
Please consider adding support for this device. I am happy to assist with any debugging or integration steps needed.

Thank you!

Koenkk, 2025-07-29:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Koenkk, 2025-08-04:
@vuvan89cz which attribute is this? (`current_heating_setpoint`?)

vuvan89cz, 2025-08-05:
> Fixed!

thank you, where can I buy you coffee?

vuvan89cz, 2025-09-02:
> @vuvan89cz <https://github.com/vuvan89cz> you can click the "sponsor"
> button here <https://github.com/Koenkk/zigbee2mqtt>, thanks!

Hello, it still shows wrong temperature after update. It should be 25, but
it still shows 2,5

vuvan89cz, 2025-09-02:
> Could you provide the debug log when it incorrectly reports this?
>
> See this <https://www.zigbee2mqtt.io/guide/configuration/logging.html> on
> how to enable debug logging.

Its working now. thank you