Koenkk/zigbee2mqtt #28101: [New device support]: Philips Hue White and Color Ambiance Xamento - single

Ruvetuve, 2025-07-29:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-xamento-inbouwspot/8720169320956

### Database entry

 "ID": 64513,                     "attributes": {},                     "commands": {                         "command1": {                             "ID": 1,                             "parameters": [                                 {                                     "name": "data",                                     "type": 1008                                 }                             ]                         },                         "command2": {                             "ID": 2,                             "parameters": [                                 {                                     "name": "data",                                     "type": 1008                                 }                             ]                         },                         "command3": {                             "ID": 3,                             "parameters": [                                 {                                     "name": "data",                                     "type": 1008                                 }                             ]                         },                         "command4": {                             "ID": 4,                             "parameters": [                                 {                                     "name": "data",                                     "type": 1008                                 }                             ]                         },                         "command7": {                             "ID": 7,                             "parameters": [                                 {                                     "name": "data",                                     "type": 1008                                 }                             ]                         }                     },                     "commandsResponse": {},                     "manufacturerCode": 4107                 }

### Zigbee2MQTT version

2.5.1

### Comments

Ref:
- https://github.com/Koenkk/zigbee2mqtt/issues/26076
- https://github.com/Koenkk/zigbee2mqtt/issues/27510

### External definition

```shell
import * as philips from 'zigbee-herdsman-converters/lib/philips';

export default {
    zigbeeModel: ['929003812801'],
    model: '929003812801',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Works as expected but no image of the device and says: unsupported. Same device as https://www.zigbee2mqtt.io/devices/929003812901.html#philips-929003812901 but this is the single lamp package of the same

Koenkk, 2025-07-29:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/philips.ts)?

Ruvetuve, 2025-07-29:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/9708