Koenkk/zigbee2mqtt #28107: Device ZG-204ZL doesn't report battery status

ustari28, 2025-07-29:
### What happened?

The device reports always 100% battery status. https://www.zigbee2mqtt.io/devices/ZG-204ZL.html

### What did you expect to happen?

It changes de status

### How to reproduce it (minimal and precise)

After some months we are receving the same status (100) of battery

### Zigbee2MQTT version

2.5.0

### Adapter firmware version

20240710

### Adapter

CC1352P2_CC2652P

### Setup

containerized

andrews6, 2025-08-17:
Same problem

Koenkk, 2025-10-06:
> After some months we are receving the same status (100) of battery

Reading this issue again, I expect that the device battery measurements are either not super accurate or that it's very energy efficient. I'm not sure if there is something to be fixed by z2m as it just send the value reported by the device. If the device would not have any reported value you would see N/A instead of 100%.